supermerill/SuperSlicer #81: [request] Manual for build from sources

userosos, 2019-07-28:
Hi! Can you make manual for build prusaslicer? I use manual from repo of the prusaslicer and it not work i hase some errors.

supermerill, 2019-07-28:
Recently, i move to visual studio 2017.
You have to cmake for "Visual Studio 15 Win64" instead of "Visual Studio 12 Win64"
I'll update the doc.

userosos, 2019-07-29:
> Recently, i move to visual studio 2017.
> You have to cmake for "Visual Studio 15 Win64" instead of "Visual Studio 12 Win64"
> I'll update the doc.

Good! I will wait doc for visual studio 2017. I am not very clever in this question so make a more or less detailed assembly manual, please.

xyyx, 2019-10-14:
You can build with Visual Studio 2019 Community
cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_PREFIX_PATH="d:\3D Printers\Slicer\destdir-64\usr\local\"

supermerill, 2019-10-16:
i've updated the doc (for windows). I may have missed something, so if it doesn't work or isn't clear, let me know.

closed 2019-11-06 by supermerill