supermerill/SuperSlicer #84: [Feature Request] Nonplanar Layers

SeverianVoid, 2019-08-23:
A slic3r implementation of nonplanar layer slicing has recently become the hot new topic and since this is the slic3r version that seems to have the most experimental and advance features figured it might be a good place for it as well.

https://tams.informatik.uni-hamburg.de/research/3d-printing/nonplanar_printing/index.php

https://github.com/Zip-o-mat/Slic3r/tree/nonplanar

https://youtu.be/km1lvuva5mI

supermerill, 2019-08-25:
I'll see if it's mergeable.

supermerill, 2019-11-26:
@Zip-o-mat said last month that "This was not intended to be merged into the master at the current state of development"

So I'll wait a bit for now. Will look again when Zip-o-mat will say that it's ready, or next year.

Zip-o-mat, 2019-11-27:
Currently I have little time to work on this. I hope that I can provide another version this year but I can definitely not guaranty it.

supermerill, 2020-10-01:
i'll see if I can add #177 as a post-process script.

theolas, 2021-04-06:
Here is another project with non planar slicing, geared towards printing 90deg overhangs without supports. 

https://xyzdims.com/2021/03/03/3d-printing-90-overhangs-without-support-structure-with-non-planar-slicing-on-3-axis-printer/

Seems like they open sourced it too.

tpimh, 2021-04-09:
@theolas This looks really impressive! Here's the repository: [Spiritdude/Slicer4RTN](https://github.com/Spiritdude/Slicer4RTN)

Right now it claims to support the following slicers: slic3r, prusa-slicer, CuraEngine{Legacy}, cura-slicer, mandoline.

jammi, 2021-04-28:
Just to ping, are there any updates to this?

arne182, 2021-06-24:
How is the progress on this feature?

supermerill, 2021-06-24:
nothing from me. I'm working on fixing bugs, and other things.
@Zip-o-mat on your side?

tpimh, 2021-06-24:
There's a related issue in Zip-o-mat's repository which was not yet linked here: https://github.com/Zip-o-mat/Slic3r/issues/34

Zip-o-mat, 2021-06-25:
> nothing from me. I'm working on fixing bugs, and other things.
> @Zip-o-mat on your side?

No, Sorry!

RaymondAndrea, 2021-10-30:
Hi, we're very interested in this and our digital fab studio would be happy to put some dev effort into this. Where could we start and how could we help?

Thanks!

tpimh, 2021-10-30:
I'm also interested in helping, but don't know where to start. Let's coordinate the efforts together!

supermerill, 2021-10-30:
I opened a discussion for it
https://github.com/supermerill/SuperSlicer/discussions/1741