supermerill/SuperSlicer #86: [BUG] Latest release missing files

reloxx13, 2019-08-26:
### Version
[stable]1.42 bugfixes

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Ender 3 Latets Marlin 2.0.x-bugfix

### Behavior
Wont start, missing file in release version:

11:32:35: can't open file '\Slic3r\Slic3r++\Slic3r++_1.42.2_190825\resources\icons\drop_to_bed.png' (error 2: Das System kann die angegebene Datei nicht finden.)
11:32:35: Failed to load image from file "\Slic3r\Slic3r++\Slic3r++_1.42.2_190825\resources\icons\drop_to_bed.png".
11:32:35: Internal error: Could not load bitmap: drop_to_bed

[image]

LaserBee, 2019-08-26:
Yep, I have the same error.

reloxx13, 2019-08-26:
i just copied the whole resources folder of the repo over the unzipped release, works now.

supermerill, 2019-08-26:
Sorry, should be fixed now.

reloxx13, 2019-08-26:
thank you, redownloaded and works fine now 👍

closed 2019-08-26 by reloxx13

M-K3y, 2021-10-03:
i have the same error with the newest versions 

20:44:28: can't open file 'D:\Downloads\SuperSlicer_2.3.56.9_win64_210906 STABLE version\SuperSlicer_2.3.56.9_win64_210906\resources\icons\.png' (error 2: Das System kann die angegebene Datei nicht finden.)
20:44:28: Failed to load image from file "D:\Downloads\SuperSlicer_2.3.56.9_win64_210906 STABLE version\SuperSlicer_2.3.56.9_win64_210906\resources\icons\.png".
20:44:28: Internal error: Could not load bitmap: 

i can download the 2 newest versions, by both the same error... the only solution for me was to rename a random png to .png to start superslicer. before i was changed some settings in the GUI tab, but i dont know what setting is the problem!?

how can i reset my superslicer config?

supermerill, 2021-10-04:
@M-K3y 
see https://github.com/supermerill/SuperSlicer/issues/1605