supermerill/SuperSlicer #88: Print Time Estimator

LaserBee, 2019-08-29:
### Version
1.42 STABLE

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anet a8

### Behavior
Previous version, the print time estimator was spot on. This version something's broken. Print time estimator is WAY off. Slic3r says 1:38, when the actual print time is 0:18.

LaserBee, 2019-08-29:
Note: I did not have the previous version, but the one before that, so I don't know if it broke for this one or the previous.

LaserBee, 2019-08-29:
Okay, I've just tried a previous version, and my print time estimate is still way off. So this means maybe I broke something. I did accidentally delete my settings and had to start again, is there something I might have missed?

supermerill, 2019-08-29:
There are a "silent mode" in the hardware tab, maybe that?

LaserBee, 2019-08-29:
I don't see any option like that. 

This is my print time estimate:

https://imgur.com/ni9h69X

But the actual print is about 2 hours.

LaserBee, 2019-08-29:
Do I need to put this at the start of the gcode? I know originally you did on the Slic3rPE print time estimator:

M201 X2000 Y2000 Z100 E10000
M203 X400.00 Y400.00 Z8.00 E50.00
M204 S1000 P400.00 T1000.00
M205 X10.00 Y8.00 Z0.30 E5.00

LaserBee, 2019-08-29:
Update: I've downloaded 1.42.0-ALPHA7.0, and it seems to work perfectly there.

LaserBee, 2019-08-29:
A bit of further investigating. I have downloaded and tried several different versions:

1.42.2_190826 BAD
1.42.1_190708 BAD
1.42.0-rc_190609 BAD
1.42.0-alpha7.0_190314 GOOD
1.42.0-alpha6.0_190221 GOOD

Now, I'm *pretty sure* that it *did* work before in a more recent one, but now it doesn't seem to!

LaserBee, 2019-08-29:
SOLVED:

After comparing the config.ini files between the non-working and working versions, I discovered that I had, somehow, changed the printer from MARLIN to REPRAP. 

I don't know how or why I did that, but I've changed it back and all is good!

You can close this. Thank you for a wonderful program.

closed 2019-08-29 by supermerill