supermerill/SuperSlicer #91: Ironing issue: small strange holes

LaserBee, 2019-08-31:
### Version
1.42 STABLE

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anet A8 (heavily modified)

### Behavior
After ironing, there are slight holes in the print, where you can almost see the layer underneath. I have tried turning the "Above Bridge Flow" up to 1.3, and I have even printed with 100% infill, and still get this. It's not everywhere, but in the badges I have phtographed you can see it's in the top left corner of each. It's not the model, as if I rotate the model it's still in the same area. 

https://photos.app.goo.gl/GjRkhv9qcKGRT7Qx9

Detail:

https://photos.app.goo.gl/8QwejEZAtc6NMF9E8

supermerill, 2019-09-02:
Maybe your filament is a bit too thin or you are under-extruding a tiny bit.
You may try to turn back the "Above Bridge Flow" up to 1.2 and put 1.05 / 1.1 to filament flow.

I also had this when i print petg in large area, but not with pla & abs. It's with pla for you?

LaserBee, 2019-09-02:
Yes, it was PLA. I will experiment with your solutions tomorrow, as I've just started an overnight print. 

It seems it only happens on some objects, maybe the object hight is just slightly too short or high?

supermerill, 2019-09-03:
The object height shouldn't be an issue.
Can you post the .3mf (project file , file -> save as) with at least two objects: a good and a bad one?

Quintox303, 2019-09-06:
I have experienced this issue too with some PLA prints. It seems in some cases there is not enough filament to fill all the gaps. I have had great success with the following ironing parameters: 
Top Solid Infill: 0.9 percentFlow
Ironing Layer: 0.3 percentFlow

I'll attach some pictures with some test prints that I've done soon

LaserBee, 2019-09-06:
I replaced the nozzle on my printer, as well as replaced the stepper motor, and it *seems* to be better. My whole printer could use a tune-up.

Quintox, where is the option to adust the ironing layer flow?

Quintox303, 2019-09-07:
Here are my results! 

So usually slic3r++ prints the top solid infill layer at 70% flow and then does the ironing layer at 30% flow. I found that this wasn't sufficient to fill everything in properly (which you can see in the pictures below), and it resulted in very rough top layers. Almost sandpaper like. I compiled another version of slic3r++ with the top solid infill layer at 90% and the ironing layer at 30% and I got awesome results. The top layer is still matte but not as rough and the extrusion lines are less visible. Here are the pictures:

[image]
Left: stock, right: 90/30
<br>
[image]
Stock slic3r++, the ironing layer extrusion is not adequately covering the layer below as you can see.
<br>
[image]
90/30 version appears to have a much more uniform ironing cover
<br>
[image]
Stock slic3r++ appears quite rough and has some holes where the infill meets the perimeter
<br>
[image]
90/30 version these holes where infill meets perimeters are gone.

My extruder was very finely calibrated before running these tests. Please note these pictures have been edited to bring out the details. 

@LaserBee unfortunately right now the ironing layer flow isn't adjustable through the software. I had to edit the parameters in the fillSmooth.hpp file and then rebuild slic3r++ to complete these tests. However that would be an awesome feature if it was configurable in slic3r @supermerill!

I will continue to do some more tests and see if I can get an even better ironing layer

supermerill, 2019-09-09:
I can easily add a "top layer flow" if needed. I just hoped it wasn't necessary. I will do that for next release.

3dprintrookie, 2019-09-09:
I have the exact issue after the latest version. A lot of gaps on the top surface that look quite ugly.

The previous version produced much better ironed top layers.

Quintox303, 2019-09-24:
I have finished my testing now. I printed about 30 test squares with different settings, comparing slic3r++ results to those from cura and adjusting appropriately. I have achieved best results with the following settings:

top solid infill layer flow: 100%
ironing layer flow: 0%
top solid infill percentWidth: 1.0
ironing layer percentWidth: 4.0 (gives a line width of approx 0.1mm)
ironing layer speed: quite low, around 40mm/s seems to be a good balance of quality and speed.
top layer fan: seems to have no impact on the ironing surface finish

Surprisingly setting the ironing layer line width so low doesn't add a huge amount of print time which is good to see. It does however make the ironing finish tonnes better. It is no longer rough to the touch, it just looks very uniform and slightly matte. 

Having the ironing layer flow set to zero also help rid the issue #96 that I was experiencing a little as well.

It would be great to have the ironing layer flow and also the flow of the layer below it adjustable. I tried adding this functionality myself but it doesn't work properly
[image]

In addition to this, I think that when ironing is enabled, the "top solid infill" extrusion width setting should impact the ironing layer only, not the layer below it. The top solid infill speed setting works this way, but not the extrusion width setting. If I set top solid infill extrusion width to 0.1mm then it will also make the solid layer below it 0.1mm.

More test pictures coming soon!

3dprintrookie, 2019-10-18:
Can you please push an update for this fix? I’ve been waiting ever since you updated the label as fixed. 

My print suffer a lot from holes on top surface with ironing

supermerill, 2019-10-18:
fixed in the lastest release. If you have holes, and you can't increase more the over-bridge flow, increase also the top flow.

closed 2019-10-18 by supermerill

3dprintrookie, 2019-10-22:
Thank you for adding the feature. @Matts-Hub what's the best setting you get with @supermerill latest update? I increased the top flow to 105% to test this out.