supermerill/SuperSlicer #93: Support enforcer not creating support gcode

toskium, 2019-09-01:
### Version
1.42.2

### Operating system type + version
Windows 10 Pro 1803

### Behavior
The support enforcer should in theory force support right under the overhang. For some reason it does not. It seems model specific since I tried to enforce support on a simple test object and that was working as expected.

Green support enforcer:
[image]

Gcode Preview:
[image]

It seems like the cylindrical shape is causing the issue. I whipped up another test part with a horizontal cylindrical extrusion and slicer++ ignores supports there as well.

Cylinder Test:
[image]

#### Project File (.3MF) where problem occurs
[support_enforcer_bug.zip](https://github.com/supermerill/Slic3r/files/3563753/support_enforcer_bug.zip)

Settings have been identical for all testing.

supermerill, 2019-09-02:
I'll look, i didn't tried this feature yet.

supermerill, 2019-10-14:
I was a side-effect (a part not well understood) of "no perimeter on bridge area" -> "fill with bridge"

closed 2019-10-18 by supermerill