supermerill/SuperSlicer #94: First line of bridge not working properly

toskium, 2019-09-03:
### Version
1.42.2

### Operating system type + version
Windows 10 Pro 1803

### 3D printer brand / version + firmware version (if known)
Hypercube Evolution (DIY)

### Behavior
* The very first parameter of a bridge is sagging down, I suppose the bridging flow is not set. (see photo) and all other bridging parameters work as expected.

* To reproduce this problem print whatever bridging parameter you want. I created a simple test object for that.

* Exptected result is a proper bridge

* Actual Result, see picture
[image]

supermerill, 2019-09-03:
This couldn't be because the fan isn't at max speed?
To test it, "disable" the fan for bridges.

toskium, 2019-09-03:
Unfortunately not, I verified fan speed as this was my first suspicion.
The fan is set to 100% with a 3 second fan startup delay.

supermerill, 2019-10-09:
I review a simple gcode and for the first perimeter (the inner one), it extrudes at 0.033 mmFil/mmMove in not-bridged area and at 0.042 in bridged area.
It's exactly the same for the second perimeter (middle one).
For the external perimeter, the amount of plastic for not-bridged section is a bit lower (0.314) but it's the same for the bridge (0.042).

So it's not the gcode fault or the bug has been repaired in the meantime.

An other explanation is a big pressure build-up in the nozzle that is release at the first bridge encountered (because it's trying to over-extrude). You may want to lower the filament extrusion ratio if it's the case.

closed 2020-11-15 by supermerill