supermerill/SuperSlicer #95: Spacing between multiple objects on buildplate

Blitzacer, 2019-09-03:
### Version
1.42.2

### Operating system type + version
Windows 10_

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S

### Behavior
When i increase the number of copies of an object, after I select "Arrange" the objects are all touching eachother, leaving no gap between them. If i print it like that, they will be fused together.
Am I missing something in the settings? 

[image]
[image]

Thanks.

PS: Any way to get the bed image like prusaslicer?

supermerill, 2019-09-03:
Reproduced.

> Any way to get the bed image like prusaslicer?

use the prusa profile from this repository, or be sure that it begin with

```
[vendor]
# Vendor name will be shown by the Config Wizard.
name = Prusa {technology}
full_name = Prusa Research {technology} Technology Printers
technologies = FFF; SLA
```

Blitzacer, 2019-09-03:
> I have it. I'll look if it's not a packaging issue. Do you use the default profile from here or did you import them?

I used a default profile from the inicial wizard.

closed 2019-10-18 by supermerill