supermerill/SuperSlicer #96: Another ironing issue: "feathering" at the edges.

LaserBee, 2019-09-06:
### Version
_Version of Slic3r++ used goes here_
1.42.2

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Anet a8

### Behavior

Ironing seems to "push" plastic over the edges, creating a very thin "feather" of plastic sticking out along the edge, perpendicular to the direction of ironing. You can see from the attached photo what I mean. I stopped that print halfway through the ironing, and you can clearly see the where the ironing stopped, there's excess plastic being pushed along. The plastic at the edges is actually hanging *over* the edge, like it's been pushed there. It's very easy to remove. 

This is with my "above the bridges" flow ration at 1, and my extruder is calibrated more or less spot on. 

https://imgur.com/DgooEV5

supermerill, 2019-09-09:
To me, it seems that you are over-extruding. Did you compare the weight of a big part (with a precision scale) and the weight the slicer give you?

3dprintrookie, 2019-09-09:
I'm having the same issues after the latest version, a bit milder than yours but ironing isn't as great as it was before.

supermerill, 2019-09-10:
I didn't touch ironing code since months.

wavexx, 2019-10-09:
On Wed, Oct 09 2019, 3dprintrookie wrote:
> https://imgur.com/a/2Wfn50K
>
> You can see it at the top of 5, and the sides of 8

This is really hard to see.
Would it be possible to have in-focus shot just a bit further away?

espwarrior, 2019-11-01:
Following this issue with curiosity. I'm currently trying to solve a minor issue with my nozzle scraping tiny amounts of plastic of the previous layer. This manifests as 1) whisks appearing where the solid infill meets the perimeter, rough 2) appearance where the direction of the infill changes (the 'U' so to speak does not look clean) and 3) a blob of PETG on the side of my nozzle that grows the longer I run the print.

I have reason to believe this is not overextrusion (or at least not entirely). Therefore, I played with settings and disabled 'Enforce 100% fill volume' which delivered much better results.

This may seem completely counter-intuitive (because I verified that disabling this option actually increases the total amount of plastic extruded), but I'm seeing no more whisks and there's no blob growing on the side of my nozzle anymore.

The included photo shows the top print with the option disabled, while the print behind has the option enabled. Both prints were not fully completed to be able to inspect the previous layer as well. The writing indicates that I calibrated and fine-tuned my extrusion and flow (@0.98) and disabled leveling fade height in this instance (M420 Z0) to exclude that as a potential cause.

[image]

Also, with the option disabled, a microscopic view of the underside of the first layer now shows that the lines of the solid infill are not fully fused anymore. This means that I should probably decrease the Z-offset a little more, which was also not apparent when I had the 'Enforce 100% fill volume' still on.

Did any of you check if you have this option enabled or that it reduces this 'feathering'?

supermerill, 2019-11-02:
Rectilinear current implementation put more plastic at the sides than at the center, with the overlap, it push much more plastic at the sides. But depending of the geometry and extrusion width, it can over-extrude or under-extrude (for the whole area).
So disabling 'Enforce 100% fill volume' can help for the sides if you are in the case where the rectilinear is under-extruding, but you will "greatly" under-extruding in the "center".

3dprintrookie, 2019-11-11:
Here is a better picture that shows feathering on all the sides using a 0.25mm nozzle (0.4mm nozzle has the same thing too)

[image]

supermerill, 2019-11-12:
Seems like it's over-extruding like hell. If there are too much plastic, the second pass will push them over the edges, i can't do anything about that.

closed 2021-03-19 by supermerill