supermerill/SuperSlicer #103: Grid and Rectiliner infil not continuous

Sidic101, 2019-10-15:
### Version
latest (1.43.0?)

### Operating system type + version
Windows 10 pro version 1903

### 3D printer brand / version + firmware version (if known)
irrelevant

### Behavior
The infil patern changes based on outer geometry, causes unwated internal bridging
[image]
PrusaSlicer 2.1 on the left, Slic3r++ 1.43.0 on the right

supermerill, 2019-10-16:
Can you drop here the .3mf (save as) ?

Sidic101, 2019-10-16:
Here you go, it's zipped because Github doesn't allow 3mf files.

[PSU.zip](https://github.com/supermerill/Slic3r/files/3734129/PSU.zip)

closed 2019-11-06 by supermerill