supermerill/SuperSlicer #104: Slicer++ Crash AddressSanitizer

yodor, 2019-10-16:
First thanks for this great work you do in this fork. Finally I have correctly printed holes :)

### Version
Slic3r-1.43.0 (1.44.0)
compiled from source

### Operating system type + version
linux gentoo 17.1

### 3D printer brand / version + firmware version (if known)
delta printer / marlin 2.0.x bugfix

### Behavior
After quiting the GUI I get a long list of errors from gcc/ASAN 

==10410==ERROR: AddressSanitizer: heap-use-after-free on address 0x60c000073a80 at pc 0x56021cffb3e0 bp 0x7fde38697940 sp 0x7fde38697930

Is the same if I slice from command line
Output result gcode is generated fine from command line

Also I get crash and immediate quit without gcode generated if I rotate an object in the plater and then slice with generate support not sure if this is related.

supermerill, 2019-10-16:
Noted. Sadly, i don't have much time to investigate these, but i keep it in my log.

> Also I get crash and immediate quit without gcode generated if I rotate an object in the plater and then slice with generate support not sure if this is related.

Can you save the projet just before clicking on "slice" ? (and zip+drop it here)

yodor, 2019-10-16:
Here it is the 3mf file zipped
[test.zip](https://github.com/supermerill/Slic3r/files/3735841/test.zip)
I found out if you use the properties boxes and rotate by inputing a number there the crash does not happen. Only if I manipulate the object rotation using the mouse - crash happens.
Let me know if I can be of help further.

supermerill, 2019-10-22:
Can't reproduce.
Can you build it in debug, to have more details about the crash?

yodor, 2019-10-27:
I can't reproduce this in debug mode. Probably something related to external widgets and themes.

closed 2019-10-27 by yodor