supermerill/SuperSlicer #107: Impossible to build for MacOS

jjjj-8, 2019-10-19:
### Version
1.43.0

### Operating system type + version
MacOS 10.12.6 (though I don't think it's relevant)

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior

Building this program is currently impossible on MacOS.
Going through this https://github.com/supermerill/Slic3r/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md works fine until ~75% through the first make command, where it hits an error and fails:

ld: can't open output file for writing: qhull, errno=21 for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [qhull] Error 1
make[4]: *** [CMakeFiles/qhull.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [dep_qhull-prefix/src/dep_qhull-stamp/dep_qhull-build] Error 2
make[1]: *** [CMakeFiles/dep_qhull.dir/all] Error 2
make: *** [all] Error 2

Now, I can't exactly read this, but I did some (hours of) searching and I _think_ the issue is due to overlapping filenames and/or directories somewhere.
Apparently, "You can't have a file and a directory with the same name on UNIX-y systems" and UNIX systems have "case-insensitive filenames"
So it looks like some issue is happening with all those various "qhull" files/folders that is causing errors.

Sources:
https://github.com/crystal-lang/crystal/issues/1540
https://github.com/crystal-lang/crystal/issues/650
https://stackoverflow.com/questions/9200521/why-does-the-name-of-a-source-file-affect-compilation
https://github.com/antirez/redis/issues/495

knoopx, 2019-10-19:
the problem is in `qhull-mods.patch`, just by replacing `/Qhull/` to something else fixes the problem

knoopx, 2019-10-19:
another problem in `deps-macos.cmake`: 

`BUILD_COMMAND` needs an `env`:

```
BUILD_COMMAND env PATH=/usr/local/opt/gettext/bin/:$ENV{PATH} make "-j${NPROC}" && make -C locale allmo
```

jjjj-8, 2019-10-20:
Okay, that all works, but after I use the `make -jN` command, it ends up failing with this output:

18 warnings and 1 error generated.
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/Fill/Fill.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
5 warnings generated.
make[1]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [all] Error 2

I don't think the warnings matter, but the error clearly does.

knoopx, 2019-10-20:
Here's what I've got so far:

https://github.com/knoopx/Slic3r/commit/565c03cf85998523993de55e75e5fd978c96eb93

compilation works, fails when linking due to missing boost symbols. Looks like a known issue on OSX. Going to research.

knoopx, 2019-10-21:
some more progress:

```
brew edit boost # then set threading=multi
brew install boost --build-from-source
```

makes most of the boost linking problems go away.
however there are still `glew` has also lots of missing symbols...

knoopx, 2019-10-21:
Another update running `brew uninstall glew` and letting cmake use the bundled one fixes the linking of glew... now im only missing:

```
Undefined symbols for architecture x86_64:
  "boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::do_assign(char const*, char const*, unsigned int)", referenced from:
      Slic3r::client::expr<std::__1::__wrap_iter<char const*> >::regex_op(Slic3r::client::expr<std::__1::__wrap_iter<char const*> >&, boost::iterator_range<std::__1::__wrap_iter<char const*> >&, char) in liblibslic3r.a(PlaceholderParser.cpp.o)
  "boost::re_detail_106600::get_mem_block()", referenced from:
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_all_states() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_startmark() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_alt() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_rep() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_char_repeat() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_set_repeat() in liblibslic3r.a(PlaceholderParser.cpp.o)
      ...
  "boost::re_detail_106600::put_mem_block(void*)", referenced from:
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in liblibslic3r.a(PlaceholderParser.cpp.o)
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::unwind_extra_block(bool) in liblibslic3r.a(PlaceholderParser.cpp.o)
  "boost::re_detail_106600::verify_options(unsigned int, boost::regex_constants::_match_flags)", referenced from:
      boost::re_detail_106600::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in liblibslic3r.a(PlaceholderParser.cpp.o)
  "boost::re_detail_106600::raise_runtime_error(std::runtime_error const&)", referenced from:
      void boost::re_detail_106600::raise_error<boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > >(boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::error_type) in liblibslic3r.a(PlaceholderParser.cpp.o)
  "boost::re_detail_106600::get_default_error_string(boost::regex_constants::error_type)", referenced from:
      boost::re_detail_106600::cpp_regex_traits_implementation<char>::error_string(boost::regex_constants::error_type) const in liblibslic3r.a(PlaceholderParser.cpp.o)
  "boost::log::v2_mt_posix::aux::unhandled_exception_count()", referenced from:
      Slic3r::ModelObject::cut(unsigned long, double, bool, bool, bool) in liblibslic3r.a(Model.cpp.o)
      Slic3r::trace(unsigned int, char const*) in liblibslic3r.a(utils.cpp.o)
      Slic3r::GUI::Plater::priv::take_snapshot(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::Plater::priv::update_after_undo_redo(Slic3r::UndoRedo::Snapshot const&, bool) in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::Plater::priv::ExclusiveJobGroup::join_all(int) in liblibslic3r_gui.a(Plater.cpp.o)
      Slic3r::GUI::generic_exception_handle() in liblibslic3r_gui.a(GUI_App.cpp.o)
      Slic3r::GUI::GUI_App::load_language(wxString, bool) in liblibslic3r_gui.a(GUI_App.cpp.o)
      ...
  "boost::system::system_category()", referenced from:
      boost::asio::io_context::io_context() in liblibslic3r_gui.a(FirmwareDialog.cpp.o)
      boost::asio::detail::conditionally_enabled_mutex::conditionally_enabled_mutex(bool) in liblibslic3r_gui.a(FirmwareDialog.cpp.o)
[... cut ...]
  "boost::re_detail_106600::cpp_regex_traits_implementation<char>::transform(char const*, char const*) const", referenced from:
      std::__1::__wrap_iter<char const*> boost::re_detail_106600::re_is_set_member<std::__1::__wrap_iter<char const*>, char, boost::regex_traits<char, boost::cpp_regex_traits<char> >, unsigned int>(std::__1::__wrap_iter<char const*>, std::__1::__wrap_iter<char const*>, boost::re_detail_106600::re_set_long<unsigned int> const*, boost::re_detail_106600::regex_data<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, bool) in liblibslic3r.a(PlaceholderParser.cpp.o)
```

jjjj-8, 2019-10-22:
Not sure I have that "missing boost symbols" issue you mentioned. When I try to build from knoopx@565c03c, (without editing boost or glew) I get this message on failure:

```
Undefined symbols for architecture x86_64:
  "Slic3r::FillGyroid::PatternTolerance", referenced from:
      Slic3r::FillGyroid::_fill_surface_single(Slic3r::FillParams const&, unsigned int, std::__1::pair<float, Slic3r::Point> const&, Slic3r::ExPolygon&, std::__1::vector<Slic3r::Polyline, std::__1::allocator<Slic3r::Polyline> >&) in liblibslic3r.a(FillGyroid.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/slic3r++] Error 1
make[1]: *** [src/CMakeFiles/slic3r.dir/all] Error 2
make: *** [all] Error 2
```

Is that the same thing?

supermerill, 2019-10-22:
> make[5]: *** [qhull] Error 1
> make[4]: *** [CMakeFiles/qhull.dir/all] Error 2
> make[3]: *** [all] Error 2
> make[2]: *** [dep_qhull-prefix/src/dep_qhull-stamp/dep_qhull-build] Error 2
> make[1]: *** [CMakeFiles/dep_qhull.dir/all] Error 2
> make: *** [all] Error 2

does your build really build qhull in the "deps" phase?

supermerill, 2019-10-22:
It seems like the "make" (from the Dependencies section) failed to build "qhull".
Can you look into your lib directory tree (maybe in /usr/local/lib) to look for a "qhull.a" (or qhull.something) ?

supermerill, 2019-11-21:
I've made some progress, having a building script here: https://github.com/supermerill/Slic3r/blob/supermerill-linux-actions/.github/workflows/ccpp_ubuntu.yml

But I don't know how to build the .dmg . I think i only have to create the CodeResources and zip the file, from what i have seem from the prusaslicer dmg.
Also, i can't check if it really works, just if it compiles.

If someone can help, this automatic build may be improved.
@domesticatedviking , advices?

AndrewFarley, 2019-11-28:
@supermerill Can you post the output of your build process somewhere, I'll gladly try it.

I just dived into trying to build this on OS-X without success.  I tried your above updated script, and ran into various errors, trying to correct each one at a time.  The deps don't install properly for me from cmake, so I tried to install them from homebrew, also with limited success and odd errors I can't seem to work around.  Each time I think I got it, some new error pops up.  It seems endless, I'm 3 hours in trying to get it to build.  If you are indeed using MacStadium (from your root README.md) you could crank out builds automatically and post them, no?  I wouldn't mind helping out in some regard if needed, I'd love to try this out and I only have macs and have experience compiling apps.

supermerill, 2019-11-28:
> If you are indeed using MacStadium (from your root README.md) you could crank out builds automatically and post them, no?

I'm not, i have to remove this icon.

> Can you post the output of your build process somewhere, I'll gladly try it.
https://github.com/supermerill/Slic3r/suites/329914526/artifacts/452241

supermerill, 2020-06-24:
If you have a new issue about macos build, maybe it's better to open a new issue.

closed 2020-06-24 by supermerill