supermerill/SuperSlicer #108: Linear advance support - modify value for infill

albert2004, 2019-10-21:
### Version
Any

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any with Marlin FW

### Behavior
Currently, it is possible to set linear advance feature value in slicer in a custom G-Code section.
This is usefull function but I think it could be tuned to make it variable.
For example, fine tuned value for linear advance makes "perfect" edges of print, but this could be not important for infill. Especially for infill where a lot of movement changes are performed (honeycomb, gyroid).
I would like to have a possibility to:
1. Set K value as an option in Slicer
2. Have a possibility to set different value or disable linear advance when printing infills (not solid).

_Is this a new feature request?_
Yes

wavexx, 2019-10-21:
On Mon, Oct 21 2019, albert2004 wrote:
> For example, fine tuned value for linear advance makes "perfect" edges
> of print, but this could be not important for infill. Especially for
> infill where a lot of movement changes are performed (honeycomb,
> gyroid).

I think it's a nice idea, esp. for grid infill.

Note that infill quality can be important though. Setting K=0 in grid
infill generally tends to produce gaps at crossings nearby walls in
large objects (this is due to the faster print speeds that can be
achieved on long lines, causing a pressure drop in the nozzle).

Increasing extruder jerk to recover speed, if possible, is a better
option. Reducing infill speed can be faster too for high K factors.

Gyroid actually performs very well with LA: it's effectively as fast as
rectilinear and/or grid, since LA kicks in only at perimeters.

Honeycomb is a different story :(

supermerill, 2019-11-06:
it's ok for you?

albert2004, 2019-11-06:
> it's ok for you?

I was able just to quickly check with your provided example but it was not working for me (always else branch was executed).
Also I think that linear advance is more specific to filament than to printer - so location under filament tab could be better.

albert2004, 2019-11-10:
OK,
verified. In the example code, there is missing asterix for regexp. After chagning into:
{if extrusion_role=~/.*InternalInfill.*/}MXXX K0{else} MXXX K20{endif}
it works.
From my perspective, functionality works as should, remains suggestion to move this (or duplicate) into filament settings as mentioned in previous post.
If not wanted to move then "issue" can be closed.
Thanks!!

supermerill, 2019-11-12:
I will maybe change the gcode generation to allow user to set the gcodes. May use a more powerful langage like lua.

closed 2019-11-12 by supermerill

wavexx, 2019-12-28:
An additional note: at least in marlin 1.1, M900 changes the global state for the planner, which means that it will change the value of the moves planned ahead of time. A change in K factor will be effective in 16/32 moves from the current one, unless M400 is issued just after M900. M400 will also bring the printer to a halt, so there's a huge trade-off to make currently...

On a small object, changing K factor for infill might actually decrease the print quality too much. Adding M400 at every change in role will probably down the print speed way more than the gain resulting by disabling LA for infill.

albert2004, 2019-12-29:
I'm using Marlin 2.x and I do not see (and hear) any delay with M900 setting. It looks like handling was improved in newer versions.

wavexx, 2019-12-29:
On Sun, Dec 29 2019, albert2004 wrote:
> I'm using Marlin 2.x and I do not see (and hear) any delay with M900
> setting. It looks like handling was improved in newer versions.

In marlin 1.1 M900 by itself will not introduce any printing delay.
What I'm referring to is the fact that the new updated K factor will
only be effective in 16+ segments from the moment you set it.

If you want to verify, use K0 for perimeters and a huge value like K5
for infill in a simple cube. Perimeters should be fast, but you should
notice very slow acceleration during infill. If the slowdown is present
also for perimeters, then you're experiencing this delay.

albert2004, 2019-12-29:
> On Sun, Dec 29 2019, albert2004 wrote: I'm using Marlin 2.x and I do not see (and hear) any delay with M900 setting. It looks like handling was improved in newer versions.
> In marlin 1.1 M900 by [...]

I was referring to this behavior. In Marlin 2.x setting is applied immediately, not after 16+ movements.

CCS86, 2021-02-17:
Is the extrusion role "perimeter" inclusive of all perimeter types?

supermerill, 2021-04-25:
not the external one.

```
Perimeter
ExternalPerimeter
OverhangPerimeter
InternalInfill
SolidInfill
TopSolidInfill
BridgeInfill
ThinWall
GapFill
Ironing
Skirt
SupportMaterial
SupportMaterialInterface
WipeTower
Mill
Mixed
```