supermerill/SuperSlicer #109: Ironing defects with 0.25mm nozzle after latest update

3dprintrookie, 2019-10-26:
Previously, there were some gaps on the top infill especially when using 0.25mm nozzle, but now, it's over extruding like crazy!

I've been at it for more than 12 hours (still going) trying to play with the new ironing parameters to fix the issue, but I either fix the gaps and ruin the edges or vice versa. I can't get a clean print at all.

Check the images, it seems like the nozzle is extruding and pushing the plastic over the edges and smudging the whole thing.

[image]
[image]
[image]

3dprintrookie, 2019-10-26:
This looks like the same feathering issue that is currently logged, but now the problem is greatly increased

supermerill, 2019-10-26:
check if the new parameter "top layer flow" is at 100%

supermerill, 2019-11-06:
With last version, do you still have the problem?
If yes you really need to send me your project file (zip it before dropping it here)

3dprintrookie, 2019-11-09:
Hey man sorry to keep you hanging, didn't notice your reply. 

I fixed it after more than a dozen tests with playing around the settings until I found out that the culprit was the **Above The Bridges** percentage. I put that to 30% to fix my issue. **Top Fill** remained at 100%.

I haven't tested your latest version. But what changed now? I am afraid that the new update will ruin my current settings that took so long to fix with tests.

supermerill, 2019-11-09:
> I haven't tested your latest version. But what changed now?

I didn't touch the "Above The Bridges".

I think i will need to do some more test with an other nozzle than 0.4.

3dprintrookie, 2019-11-11:
Got more pictures of the surface, not really ideal even if I put the ATB to 35%

[image]
[image]
[image]
[image]

supermerill, 2019-11-12:
can i have your .3mf (project file, sipped and dropped here) ?

3dprintrookie, 2019-11-29:
There you go

[raptorslicer.stl.zip](https://github.com/supermerill/Slic3r/files/3904749/raptorslicer.stl.zip)

supermerill, 2019-11-29:
@3dprintrookie it's the stl, not the 3mf. Can you click on "save as" in slic3r++ just before/after slicing your object and sending the zip of this file instead?

Without that, i don't know what settings you used and so i can't replicate.

Nyshan, 2024-01-02:
@3dprintrookie @supermerill Is this still a problem? This issue is now four years old and this question: https://github.com/supermerill/SuperSlicer/issues/109#issuecomment-550326357 was never properly answered.

closed 2024-01-16 by supermerill