supermerill/SuperSlicer #110: 3D Honeycomb infill sliced as bridge feature

yodor, 2019-10-27:
### Version
Slic3r-1.43.0 (1.44.0)
compiled from source

### Operating system type + version
linux gentoo 17.1

### 3D printer brand / version + firmware version (if known)
delta printer / klipper 0.80

### Behavior
3D Honeycomb infill set as bridge feature.

[image]

#### Project File (.3MF) where problem occurs

It happens on every file I tried 3d honeycomb infill

supermerill, 2019-10-29:
Done. I removed the "force bridge" feature, so it can't happen now.

wavexx, 2019-10-29:
Can I ask if this is a bug that comes from upstream?
I'm not at home to check, but I'm quite interested to know why some infills force bridging and/or use bridge flow.

supermerill, 2019-10-29:
It's an old "feature". As the infill isn't printed exactly on top of an other layer, someone thought that bridged extrusion will print better.

But it's only true if layer_height > nozzle_width/2

also, bridging can be very problematic with a small layer height: if you have only 1 or 2 layer of sparse infill between solid layer, slic3r doesn't think twice and will create a bridging layer, which will greatly overextrude and make the print ugly.

That's something i have to correct (already implement a volume check for x and y but not z, it's much more difficult) ... later, when i'll have the time or when many people hit this wall and report.

closed 2019-11-06 by supermerill