supermerill/SuperSlicer #112: Potential issue with 'filling horizontal gaps on slopes' option

espwarrior, 2019-10-30:
### Version
2.1.44

### Operating system type + version
Windows 10 Professional

### 3D printer brand / version + firmware version
N/A

### Behavior
Enabling the option 'filling horizontal gaps on slopes' generates circular perimeters on layer 3 (@0.60mm) to cover open spaces in my model (inset text facing the build plate), instead of generating the expected solid and bridge infill. The criteria for a slope don't seem to be present, but I could be misinterpreting what this option is intended to achieve. I think the option was enabled due to imported settings from a previous version of Slic3r++ 1.42.2.

_Please also note that while 'Interior brim width' is zero, there are circular fragments of brim within the perimeters of some letters in the first layer. This might be a separate issue however._

With the 'extra perimeters' option enabled:

[image]

[image]

With the 'extra perimeters' option disabled (which is the expected result):

[image]

#### Project File (.3MF) where problem occurs

[Tube Cap.zip](https://github.com/supermerill/Slic3r/files/3791369/Tube.Cap.zip)

supermerill, 2019-10-30:
I think the first is a duplicate of #46. It's not really a bug but an inconvenience: as long as there are a detected hole, it add a perimeter. As your text / hole go to the middle, it adds perimeter to the middle. The algorithm should be changed/ modified.

The second problem is a more serous bug, can you paste here your project file (.3mf in a zip) ?

espwarrior, 2019-10-31:
Ah, yes, #46 looks like the same issue indeed. It was problematic because the holes were covered by regular perimeters instead of using bridges, causing a bit of sagging. Apologies, I had included the [link](https://github.com/supermerill/Slic3r/files/3791369/Tube.Cap.zip) to the 3mf already, but not in a logical and clearly visible spot.

supermerill, 2019-11-01:
I understand now why it creates a bit of brim inside: it's because i changed the logic to allow islands inside holes, and here you have the D inside the P that trigger the brim.

I'll add an other setting to enable/disable brim inside holes.

espwarrior, 2019-11-01:
Cool! At first, to me it seemed like one big oversized brim in the wrong place, but now I notice that there are indeed actually five separate 4mm brims for the islands within a, e, p, o which have overlap. I'm glad you found the cause!

closed 2019-11-06 by supermerill