supermerill/SuperSlicer #113: Unexpected behavior for extrusion width

espwarrior, 2019-11-01:
### Version
2.1.44

### Operating system type + version
Windows 10 Professional

### 3D printer brand / version + firmware version
N/A

### Behavior
I use a 0.40mm nozzle on my printer and slice any model with 0.20mm layer height. Under the **Print Settings** &rarr; **Width & Flow** &rarr; **Extrusion Width**, I have all values set to 0:

[image]

**Print Settings** &rarr; **Perimeters and Shells** &rarr; **Recommended object min wall thickness** then calculates a wall thickness of 0.80mm for two perimeters. This corresponds with flow math using a 0.42mm external perimeter and a 0.45mm regular perimeter:

[image]

The corresponding tooltip explains that if the value of 'external perimeter width' is 0, the 'default extrusion width' is used if that value is set. So if I explicitly set a default extrusion width of 0.45mm, I expect that this thickness applies to the external perimeter as well, which should result in a wall thickness of 0.86mm for two perimeters.

[image]

However, the actual wall thickness is still 0.80mm.

Only when I explicitly set the external perimeter width to 0.45mm:

[image]

The wall thickness for two perimeters changes to 0.86mm:

[image]

So there seem to be two issues at play:

1. A set value for 'Default extrusion width' is not used for 'External perimeter width' as should happen per the 'rule set' described in the tooltip. I'm also seeing evidence that one or more of the following parameters 'First layer', 'Perimeters', 'Infill', 'Solid Infill', 'Top Solid Infill' and 'Support Material' are affected by this as well.
2. The standard 'External perimeter width' of 0.42mm that Slic3r++ uses does not correspond with the description of the tooltip, which states to use 112.5% of the nozzle size (0.45mm). I see in the C++ code that Slic3r++ uses a factor of 105%, which is different from PrusaSlicer (which does use 112.5% for external perimeters). I'm expecting here that the correct factor is stated in the tooltip and documentation, so this is more of a cosmetic issue.

#### Project File (.3MF) where problem occurs
Every project I tried so far.

supermerill, 2019-11-02:
I think I'll rework these settings, it's all over the places, too much spaghetti code.

Also, the tooltip doesn't seems right, have to double-check it.

Will update the tooltips.

espwarrior, 2019-11-05:
Wow, thanks! I looked at some of the changes you already made and it seems that much more code is involved than I anticipated...

Matts-Hub, 2019-11-08:
The 2.1.45.0 release seems to give incorrect values in the recommended object wall thickness tooltip. I am seeing values that are double the (correct) values from the previous release.

supermerill, 2019-11-08:
Previous result depends if "thin wall" is activated. The new one only give the result for "normal walls" (perimeters), maybe that's why?
If it's not that, what number do you see for how many perimeters and extrusion width?

Matts-Hub, 2019-11-08:
I have tried with thin walls enabled and disabled and nothing changed. Interestingly in the previous version I found that the values remained the same if thin walls was enabled, but rather than showing values for 2, 4, 6 or 8 perimeters, it shows all values from 1-8. Is this what you mean?

With perimeter width set to 0.48 and 0.2mm layer height, I have values of 1 perimeter: 0.92mm, 2 perimeters: 1.83mm, 3 perimeters: 2.71mm, 4 perimeters: 3.58mm. This works out to be double of what the previous version shows

supermerill, 2019-11-08:
When you set "1" in the "perimeters" setting, you need to have at least a 0.92mm wall for it to be extruded by the perimeter.
When you set "2" in the "perimeters" setting, you need to have at least a 1.83mm wall for it to be extruded with the two perimeters.

It's double because you need the space for the perimeter to go back on the other side.
Before it was more "extrusion" than "what you can put in the perimeter setting".
1 periemter = 2 external perimeter extrusions
2 perimeters = 2 external perimeter extrusions and 2 internal perimeter extrusions

But it's maybe more confusing now? Can you help me to find a better (and concise) wording?

Matts-Hub, 2019-11-10:
Got it. That makes perfect sense now that you explain it. I think the wording of it should be similar to how you have it now. Something like "Suggested minimum object thin wall thickness for x layer height and x perimeters" would be good. 

> It's double because you need the space for the perimeter to go back on the other side.

Understood.  I think your current implementation of it is better

closed 2020-01-10 by supermerill