supermerill/SuperSlicer #114: Need to move the nozzle up at end of printing on purge block during retract filament change.

TNDavid, 2019-11-04:
### Version
_Version of Slic3r++ 1.43.0

### Operating system type Windows 10-1903

### 3D printer brand / version + firmware version (if known)
Prusa MK3, firmware 3.7.2-2363

### Behavior
At end of printing on purge block, nozzle remains in contact during retraction and filament change.  This  "cooks" the first layer onto the PEI print surface, even when using Windex with PETG.  The print removes well everywhere except where these spots have been cooked onto the surface!  It requires using a blade or spatula to get these spots removed, at danger of pulling PEI up off steel sheet.

_Is this a new feature request?_    YES

supermerill, 2019-11-04:
Problem is, it's a bit dangerous to move up because of a potential oozing, creating a feature taht can be pushed by the extruder later, creating problems.
Maybe it can continue to zigzag like it does when it cools the filament?

TNDavid, 2019-11-04:
That would certainly be better! Sitting in one spot so long is a real
problem.