supermerill/SuperSlicer #115: Unnecessary "Prusa Dance" at final filament retraction on multi materiel print.

TNDavid, 2019-11-04:
### Version
_Version of Slic3r++1.43.0
 
### Operating system type + version
Win 10-1903
### 3D printer brand / version + firmware version (if known)
Prusa MK3, firmware 3.7.2-2363

### Behavior
When multi material print finishes, the last filament used retracts out of extruder, then i          t does the "Prusa Dance" before returning the filament to the MMU2.  It should just return the filament to the MMU2 the same way it does for all the other filament changes.  This must be the Prusa print end moves still being done where not needed.

_Is this a new feature request?_ Yes

supermerill, 2019-11-04:
All of that it inside the "end gcode" of the prusa config bundle. Unless mine is out of sync with the prusa one, it's not really my responsibility. You can create your one and share it, or find a better one.

TNDavid, 2019-11-04:
I will look at this further. Thanks for your replies!

closed 2019-11-06 by supermerill