supermerill/SuperSlicer #116: Thin walls not being sliced properly

Matts-Hub, 2019-11-11:
Note that this has nothing to do with the "thin walls" option. 

Here I'm referring to walls in models that are around 0.8mm thick or less. Slic3r++ will not slice them and leaves gaps, but prusa slicer handles it perfectly. It seems like prusa slicer is able/willing to squeeze perimeters closer together than what slic3r++ is doing. Slic3r++ just rejects the areas and ignores them. I've seen this in many models but its just been pointed out to me that prusa slicer handles them well while slic3r++ does not. I'm using the latest release of slic3r++.

I have tried playing with the following options:
Switching thin walls on: This fills the area, but the surface is inconsistent between where it switches from two perimeters in the wall to one. See image below.
Gap fill: no difference
Lowering extrusion width: Did work but is not a solution. The slicer should be able to handle it, as prusa slicer does.
Ensure vertical shell thickness: no difference
Extra perimeters: no difference

Slic3r++ Demo (3MF file is attached)
[image]

Prusa Slicer Demo (Same profile, imported from 3MF file)
[image]

Slic3r++ with thin walls enabled
[image]

And another demo here, slic3r++:
[image]

Prusa Slicer:
[image]

[thinwalldemo.zip](https://github.com/supermerill/Slic3r/files/3831835/thinwalldemo.zip)

supermerill, 2019-11-12:
I removed the bit of code that allow perimeters to be grossly over-extruded by extruding them on top of each other.

Please use thin wall option.

ogland, 2019-11-12:
Hi @supermerill ! It was me who discovered this. I think the general consensus in my bubble is that thin walls (detect thin walls) creates unnecessary seams, retractions, unretractions, travels and inconsistent walls. In an actual print with the default way of doing it in Prusa Slicer I have never noticed any overextrusion in areas like this. Not big enough to warrant extra seams, retractions, unretractions, travels and inconsistent walls.

Perhaps create an option to use the default way of slicing walls like PrusaSlicer does it? With "detect thin walls" option OFF that is.

I'd like to quote what @Matts-Hub wrote here
> Switching thin walls on: This fills the area, but the surface is inconsistent between where it switches from two perimeters in the wall to one.

Thank you for your amazing work.

supermerill, 2019-11-12:
> in Prusa Slicer I have never noticed any overextrusion in areas like this

look at the last picture, you can clearly see at the bottom of the curve that the two perimeters are printed almost at the same place, so there are almost two times more plastic than needed here, and this plastic will go away and will create a thicker "wall" (and maybe a bit uglier vs a model with a thicker one).

> I think the general consensus in my bubble is that thin walls (detect thin walls) creates unnecessary seams, retractions, unretractions, travels and inconsistent walls
> Perhaps create an option to use the default way of slicing walls like PrusaSlicer does it? With "detect thin walls" option OFF that is.

I can, it's easy. Note that it's bit like saying "please increase model thickness where it's between [width] and [2*width] mm to [2*width] mm to ensure these features are printed with normal perimeters".

ogland, 2019-11-12:
I'm gonna try printing it both ways and see how the results look instead of just discussing theory further. But I haven't ever been bothered by overextrusion in thin walls with the exception of when they're very short in x/y, like 6-10 mm.

I'll return with some findings.

supermerill, 2019-11-12:
Just a little note: the posted model is correctly sliced with a 0.42 mm external perimeter.

closed 2020-01-10 by supermerill