supermerill/SuperSlicer #117: Extremely thick top layer

Jademalo, 2019-11-11:
### Version
2.1.44

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Custom Prusa i3 clone running Marlin 2.0

### Behavior
* _Describe the problem_
When printing with Rectilinear top solid infill, I get severe overextrusion resulting in an extremely rough top surface.
* _Steps needed to reproduce the problem_
Literally just slice anything
* _Expected Results_
Relatively smooth and flat top, like I get in PrusaSlic3r
* _Actual Results_
Extremely rough top surface

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[test_plug.zip](https://github.com/supermerill/Slic3r/files/3833010/test_plug.zip)
[config Slic3r++.zip](https://github.com/supermerill/Slic3r/files/3833012/config.Slic3r%2B%2B.zip)

I have absolutely no idea what is causing this, but it's driving me mad.
I was using PrusaSlicer without any issues, and swapped to Slic3r++ to take advantage of the hole size feature. I simply imported my config bundle, and ever since then my top layers have been substantially overextruded.

This is not an issue on any other layer, and I've calibrated my Linear Advance K factor, my extruder's steps/mm, and everything else. Everything extrudes fine except for top infill.

supermerill, 2019-11-12:
I searched but i didn't find anything wrong.

Can you post 
 * your .3mf/config from prusaslicer 
 * your gcode from prusaslicer
 * your gcode from slic3r++

Jademalo, 2019-11-21:
Spent the last week trying to get to the bottom of it, finally worked it out. It was a series of insane coincidences that made me think it was the slicer's problem.

Basically, there was a loose connection to the stepper motor of the extruder. Whenever I had been calibrating the extrusion length, it had been in such a position that it wasn't correctly connected, and was underextruding. I would then tweak the steps/mm to compensate. 
Whenever I ran a normal print, due to the position of the hotend during the top layer, it would just move the wire enough to make the connection to the extruder strong again, resulting in overextrusion.

When I did the test where I used prusaslicer, I had just moved the printer to a different table. In moving it, I'm assuming the connection had been wiggled enough that it again didn't connect correctly, and so underextruded for that print. When doing a test with slic3r++ immediately afterwards, it seems like the connection was being made correctly again for whatever reason.

After re crimping the connector, both slicers are giving me the correct output, so I've recalibrated the steps/mm and I'm no longer having overextrusion. 

Sorry for the waste of time, and thanks for the work you're doing on this slicer!

closed 2019-11-21 by Jademalo