supermerill/SuperSlicer #121: [Feature Request] Adding Tubulane infill

Gosga365, 2019-11-18:
The 'bullet proof' printing done by this group:
https://www.google.com/amp/s/www.techradar.com/amp/news/3d-printers-have-made-bulletproof-cubes-vests-and-tank-armor-next

used a printed model of a tubulane structure printed in PLA to achieve high strength parts. I was thinking this would be an interesting pattern to implement as a new infill pattern for higher strength printing. I believe you worked on the gyroid infill development before and was hoping you would be able to give me some pointers to work on developing a new infill pattern.

supermerill, 2019-11-19:
For what i can read, it seems identical to 3D honeycomb.
If you can find a paper that better describe the structure, maybe we can spot a difference, but for now i can't see one.

Chiibit, 2020-01-11:
Does [this](https://onlinelibrary.wiley.com/action/downloadSupplement?doi=10.1002%2Fsmll.201904747&file=smll201904747-sup-0001-SuppMat.pdf)  have what you'd need? The [article](https://onlinelibrary.wiley.com/doi/abs/10.1002/smll.201904747) it's supporting is behind a paywall. If I, personally, had a valid reason to contact one of the authors directly, since they don't get any money from these sites, I'd do that, but I wouldn't be able to answer any questions if asked.

One [3D printing site](https://3dprintingindustry.com/news/the-right-3d-printed-lattice-can-stop-a-bullet-in-its-path-165881/) describes it as "tubulanes are complex structures based on fullerene tubules made from carbon. Fullerene tubules themselves, from the buckminsterfullerene or “buckyball“, consist of molecules that form a closed cylinder."

JS3910, 2020-01-18:
I'm going to assume it's just the normal level of tech/science reporting, but the claims in the article makes no sense, like "cubes of ‘tubulane-like ultrahard polymers’ which are lightweight and full of holes, but as hard as diamond.". They don't say what actual polymer this is but obviously there are no plastics that have the hardness of diamond, nor can the shape of a structure affect the hardness of the material. It seems to be a good structure for dissipating the energy of an impact but that's not directly related to neither hardness nor strength of the material.

liftbag, 2020-02-02:
> For what i can read, it seems identical to 3D honeycomb.
> If you can find a paper that better describe the structure, maybe we can spot a difference, but for now i can't see one.
> 
Looking around, it also seems to me to be 3D honeycomb.
In this regard, I must note that the 3D honeycomb is only partially drawn, alternating the angles at each layer. This makes the structure light and quick to perform, but not exactly solid, as can also be seen from the preview.
Maybe you could insert a new 3D full honeycomb option, where at each layer the hexagons are executed entirely.

[image]

gringer, 2023-03-10:
@liftbag fixed [here](https://github.com/supermerill/SuperSlicer/issues/506#issuecomment-1330399559), it's just not in the main code yet. I don't understand why.