supermerill/SuperSlicer #122: Incorrect nozzle set temperature after tool change

fromanza, 2019-11-23:
### Version
V2.1.44

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3 MMU2S

### Behavior
Firstly, tips are FINALLY perfect!!! No strings :D

The issue is using PET on EXT1 and PVA on EXT5. After the temp change (using hot end fan) the filament rams and then extracts. Then the Gcode calls for the previous temperature to be set. This is incorrect. The NEXT filament temperature change should be set otherwise we end up printing PVA at the PET temeprature which of course burns. Similarly, the PET section set to the PVA temeprature on so on.

[Skinny.zip](https://github.com/supermerill/Slic3r/files/3882384/Skinny.zip)

;--------------------
; CP TOOLCHANGE START
; toolchange #2
; material : PVA -> PET
;--------------------
M220 B
M220 S100
; CP TOOLCHANGE UNLOAD
G1  X157.413 Y121.017 
M900 K0
G1 F2067
G1  X166.026 E0.3990
G1 F2102
G1  X174.785 E0.4057
G1 F2137
G1  X183.688 E0.4124
G1 F2171
G1  X192.735 E0.4191
G1 F2206
G1  X201.928 E0.4258
G1 F2258
G1  X211.338 E0.4359
G1 F2363
G1  X215.913 E0.2119
G1  Y121.677 F7200
G1 F2363
G1  X210.644 E0.2440
G1 F2606
G1  X199.787 E0.5029
G1 F3057
G1  X187.048 E0.5901
G1 F3735
G1  X171.486 E0.7209
G1 F4604
G1  X157.413 E0.6519
G1  Y122.337 F7200
G1 F4604
G1  X162.522 E0.2366
G1 F5559
G1  X185.684 E1.0729
G1 F6445
G1  X212.538 E1.2439
G1 F7140
G1  X215.913 E0.1564
G1  Y122.997 F7200
G1 F7140
G1  X189.539 E1.2217
G1 F7609
G1  X157.836 E1.4685
G1 F7852
G1  X157.413 E0.0196
G1  Y123.657 F7200
G1 F7852
G1  X189.707 E1.4959

M106 S127 ;Part fan on to cool hotend
M109 S175 ;SKINNYDIP TOOLCHANGE WAIT FOR TEMP NORMAL MODE
M106 S0 ;Fan off

G1 E-15.0000 F7200
G1 E-24.5000 F1200
G1 E-7.0000 F600
G1 E-3.5000 F360

*********THIS LINE SHOULD BE SETTING THE NEXT FILAMENT TEMPERATURE**********
M104 S205  ;RESTORE PRE-TOOLCHANGE TEMP NORMAL MODE
*******************************************************************************************

G1  Y124.317
G1  X157.413 E20.0000 F388
G1  X189.707 E-20.0000 F323
G1  X157.413 E20.0000 F258
G1  X189.707 E-20.0000 F194
; SKINNYDIP START
G1 E45.0000 F1980
G4 P50
G1 E-45.0000 F4200
G4 P0
; SKINNYDIP END
G1 E-35.0000 F2000
G1  Y124.237 F2400
G4 S0
; Filament-specific end gcode

#### Project File (.3MF) where problem occurs

fromanza, 2019-11-23:
Perhaps your code above is not necessarily incorrect. If it stays as is (which is useful/necessary for the next filament to purge out the old filament ni the nozzle but at some stage during the purge the new filament temperature should be set. I will try this.

Eboman123, 2019-11-25:
I can concur.  The print temps seem to be flip-flopped.  Filament A will change/purge correct, but then temp will change to the Print Temp for Filament B while printing with Filament A.  Then change from A to B is successful, but the print temp when printing with filament B is the temp for filament A.  

I have not tried 3 or more different filaments to see how using more than 2 filaments operates, but I have duplicated the error reported.

fromanza, 2019-11-26:
One work around is to add the filament temperature to the "filament start" section in Slic3r.
This temp change setup to me seems the only reliable & logical way to remove strings. The problem now is that if ramming is used, ramming completes and leaves residual nozzle pressure which then oozes out and creates a blob on the wipe tower while the nozzle cools down in preparation for extraction. My only solution so far was to remove ramming altogether to avoid this. Ideally the print head should be moving up and down the wipe tower depositing any remnant filament while the nozzle cools.

fromanza, 2019-12-05:
This took a while for me to figure out but Gcode apparently doesnt always run sequentially when buffers are involved IE movements will always run sequentially but temperature settings may not. There was an issue opened with slic3r a while back talking about this: 
[https://github.com/prusa3d/PrusaSlicer/issues/1339](url)
So it seems adding a G4 S0 before the filament temperature change might be necessary:
_G1  Y148.078 
M900 K30; Filament gcode
**G4 S0**
M104 S205  ;RESET THE FILAMENT PRINT TEMEPERATURE_

It will also be necessary to do this before the restore pre-toolchange temp. What I have realized is that the restore pre-tool change temp is to allow the old filament to be purged out on the new filament load. As the new filament is loaded the above code sets the new temperature. Should be fine for decreasing temperatures to use M104 but when you have a PVA going to PET then there might be extruder grinding as the nozzle wont be hot enough IE M109 might be required.

_M106 S255 ;Part fan on to cool hotend
M109 S210 ;SKINNYDIP TOOLCHANGE WAIT FOR TEMP NORMAL MODE
M106 S0 ;Fan off
G1 E-15.0000 F9000
G1 E-24.5000 F1200
G1 E-7.0000 F600
G1 E-3.5000 F360
**G4 S0**
M104 S245  ;RESTORE PRE-TOOLCHANGE TEMP NORMAL MODE
G1  Y147.908
G1  X96.470 E20.0000 F1755_

supermerill, 2020-01-10:
i just correct a bug about somethign similar, i didn't check your bug yet, but it's maybe resolved.

TNDavid, 2020-04-10:
I did not find this post until after I made mine.  (#195).  I does not work correctly when different materials are used with string reduction (Skinnydip).   Sure hope getting it back in the light will bring some needed attention to it!

closed 2020-10-01 by supermerill