supermerill/SuperSlicer #125: Bridge & Solid Infill problems

Alphaelectric, 2019-12-08:
### Version
Slic3r++ 2.1.44

### Operating system type + version
Windows 10 64Bit

### 3D printer brand / version + firmware version (if known)
Ender 3 + Marlin 2.0 Bugfix 

### Behavior
Slic3r++ creates infill that is excessively hard to print and may lead to longer prints (see attached images). 
The problem manifests itself in areas that require a lot of bridging. Both PrusaSlicer 2.1 and Cura 4.1 create a uniform ceiling on top of the areas to bridge. Slic3r++ creates a lot of different patches with bridge infill.

The second part of the problem is the solid layer that is printed on top of the bridge infill. As the screenshots show, PrusaSlicer 2.1 as well as Cura 4.1 create a uniform layer that is easy to print. Slic3r++ does not create such a uniform pattern. Instead it creates lots of small islands that a difficult to print and take far longer than required. 

The actual result was a failed print due to collision. 

Screenshots are taken from Slic3r++ and PrisaSlicer 2.1 for comparison of the same area and layer height. 

[image]
[image]
[image]
[image]

supermerill, 2019-12-09:
1) uncheck "more perimeters"
2) increase "margin: Default"  (by 2/3)

Alphaelectric, 2019-12-10:
Thanks for your help. Unchecking "Extra Perimeters" makes it better, however, some pockets still remain. I cannot find the value for "margin". What exactly are you referring to? I am using the english language GUI,
[image]
[image]

supermerill, 2019-12-10:
infill -> Anchor solid infill by X mm -> default is for growing the top layer (and all solid below). Bridge is for growing the bridge layer and all solid layer above. So i think you want to increase the bridge anchor a bit here (not the default)

Alphaelectric, 2019-12-11:
That did it! At 440% the sliced image was almost identical to the other results. I will do test prints to verify.

closed 2019-12-12 by supermerill