supermerill/SuperSlicer #127: Link to github actions is broken

JS3910, 2019-12-12:
After trying to build on Mac a few times over the months but failing horribly I was so excited to see there's an automated build now, but tragically, the github actions link on the front page is broken with a 404 error!

supermerill, 2019-12-12:
I'm all for that but i really don't know how to compile on mac.
[here is my command list to build on mac](https://github.com/supermerill/Slic3r/blob/master/.github/workflows/ccpp_mac.yml)

basically, after building deps, i do
```
cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local" -DSLIC3R_STATIC=1
make slic3r
```
It's very very simple. Everything is taken care of in the cmakefiles. 
How to say to build it for an other sdk version? 
I have a CMAKE_OSX_DEPLOYMENT_TARGET variable
Printed during the build, process it's set to "default" and after that to "10.15"
I think I can try to add -DCMAKE_OSX_DEPLOYMENT_TARGET="10.14" to the cmake command.

Big question: which version should i target, and when i will have to change?

JS3910, 2019-12-12:
The link works now, weird.

I'm not enough of a programmer on any platform to give an authoritative answer, but the official PrusaSlicer instructions say the following:

> By default PrusaSlicer builds against whichever SDK is the default on the current system.
> 
> This can be customized. The CMAKE_OSX_SYSROOT option sets the path to the SDK directory location and the CMAKE_OSX_DEPLOYMENT_TARGET option sets the target OS X system version (eg. 10.14 or similar). Note you can set just one value and the other will be guessed automatically. In case you set both, the two settings need to agree with each other. (Building with a lower deployment target is currently unsupported because some of the dependencies don't support this, most notably wxWidgets.)
> 
> Please note that the CMAKE_OSX_DEPLOYMENT_TARGET and CMAKE_OSX_SYSROOT options need to be set the same on both the dependencies bundle as well as PrusaSlicer itself.
> 
> Official Mac PrusaSlicer builds are currently built against SDK 10.9 to ensure compatibility with older Macs.
> 
> Warning: XCode may be set such that it rejects SDKs bellow some version (silently, more or less). This is set in the property list file
> 
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Info.plist
> To remove the limitation, simply delete the key MinimumSDKVersion from that file.

I don't know if it's difficult and/or complicated to install older SDK's for building though, I'd say targeting 10.14 will be fine for those reluctant to upgrade and should presumably be fine for many years judging by Prusa official builds still building for 10.9 (from 2013). Extra excitement gets added from the fact that I don't know what if anything in this document is accurate, considering I never managed to get it to actually build!

Source:
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md

supermerill, 2019-12-12:
I'll try with 10.14
No promise, there are a high probability of failure, because it's not installed anymore (and they said that we have to use our own computer if it's something we want).

supermerill, 2019-12-12:
It seems to have worked.

JS3910, 2019-12-14:
[image]

It starts and loads the interface fine but instantly throws this error, then quits.

closed 2019-12-14 by JS3910