supermerill/SuperSlicer #129: Mac version using low resolution graphics.

JS3910, 2019-12-14:
### Version
2.1.46, nightly build for Mac linked in the other issue.

### Operating system type + version
Mac OS 10.14

### Behavior
Slic3r++ renders graphics with low DPI, and doesn't use Retina graphics - PrusaSlicer works fine in this regard. The program works fine but it does cause everything to be fuzzy.

[image]

supermerill, 2019-12-18:
> it doesn't have an icon on the application package.
> It also doesn't add a shortcut to the launchpad.
>I suspect all these issues are as simple as packaging flags or something like this.

> I'm not knowledgeable enough about it to actually make a useful comment :).
Sadly I'm not better, I don't know how to create a good macos package.

> PrusaSlicer seems to just disable dark mode for itself across the entire app.
Yes, they added some commit on that after i merged. These will be merged in the next version, or a better fix from them, hopefully.

> Slic3r++ renders graphics with low DPI, and doesn't use Retina graphics - PrusaSlicer works fine
This may be the same problem as the dark mode: the macos sdk has evolved but the gui library used (wxwidget) has not be updated for it yet.

JS3910, 2019-12-18:
I have sleuthed into the internet and have labored to produce the following information:

The mystery packaging magic missing to fix almost all of this is a file called info.plist, which is put into the Contents folder in the app package. This is an XML file with a lot of info, including the app name and icon file. I just opened and edited the one from the PrusaSlicer package a slight bit (most importantly changing the bundle identifier or Mac OS gets really confused) and it works fine, high resolution graphics and all. 

The official documentation for this file is here:
https://developer.apple.com/library/archive/documentation/General/Reference/InfoPlistKeyReference/Introduction/Introduction.html#//apple_ref/doc/uid/TP40009248-SW1

The PrusaSlicer github has this file which I suspect is somehow used to generate this file:
https://github.com/prusa3d/PrusaSlicer/blob/master/src/platform/osx/Info.plist.in

I did not manage to make the icon work yet but here's some notes from testing:

I took the icns file for the original Slic3r from here to use:
https://github.com/slic3r/Slic3r/blob/master/var/Slic3r.icns

For a custom icon, generating a proper icns seems somewhat sensitive (online converters I tried did not work) but these are the best guidelines I found online, though I didn't try them and it also requires a Mac to run the converter program. I don't think you actually have/use a custom icon though, no?

https://elliotekj.com/2014/05/27/how-to-create-high-resolution-icns-files/

That said, I put this into the resources/icons folder and changed the icon file name in the plist, but it hasn't worked yet. A very minor issue though, the big one is the high resolution support and that's fixed by the above.

JS3910, 2019-12-18:
I tried to opt out of dark mode by adding this to the plist:

<key>UIUserInterfaceStyle</key>
<string>Light</string>

Which the internet suggests should work, but it didn't work for me. If it's fixed in code then it doesn't matter anyways though.

supermerill, 2019-12-19:
can you test the lastest nightly? https://github.com/supermerill/Slic3r/suites/365217277/artifacts/752669

JS3910, 2020-01-18:
Sorry, got distracted by all this holiday business. This version works fine, it renders in high resolution properly, gets added to launchpad and doesn't get munged by dark mode. The icon isn't working but that's not an overwhelming issue.

closed 2020-01-18 by supermerill