supermerill/SuperSlicer #130: Colour Change colour selector

LaserBee, 2019-12-14:
I don't know if this is the place to leave constructive criticism...

I appreciate what you've done with the little colour-change button, where you can inject a colour change gcode in at specific layers. It's neat that you can now pick what colour you want each section. 

BUT, I don't know how good a feature it actually is. It is extra clicks, and I'm not sure it's worth it. Here's how the workflow used to be:

1) Slice the object. 
2) Go to first colour change layer
3) Hit the +
4) Go to next colour change.
5) Hit +
6) repeat. 

Now here's how it is:

1) Slice the object. 
2) Go to first colour change layer
3) Hit the +
4) Pick a colour
5) Hit OK
4) Go to next colour change.
5) Hit +
6) Pick a colour
7) Hit OK
8) Repeat

It's an extra two clicks for every colour change. I don't think there's *much* use in being able to pick your own colours, it's neat, but not very useful, and not really worth the extra two clicks every time you want to add one in (and some of my designs have 6 or 7 colour changes!) 

Anyway, if there was a way to disable this feature, and just have it pick the colours at random like the previous version, it would be great!

Absolutely love the program though, and appreciate all the effort you put into it! Do you have a patreon?

supermerill, 2019-12-16:
Sorry. It's a thing that get along with the modification of the prusaslicer.
As my last beta is based on their dev branch (to correct a bug), these are things that can appear.

You have to wait the release of their 2.2 alpha, test it, and then post on their github. I think it's best to wait their release to see exactly what and why they change it.

LaserBee, 2019-12-16:
Ah, okay, so I should ask them to change it. Thanks very much!

Do you have any way to donate to you? I really appreciate the work you put into this!

LaserBee, 2020-02-24:
Hi there! I have noticed that all the issues I had (this one, and the issue with the spacemouse) have been fixed in the latest Prusaslicer release. I'm not trying to rush you or anything, I was just wondering what kind of timescale it normally takes to get it into your releases? 

You're doing a fab job here anyway. Do you have Patreon?

supermerill, 2020-02-24:
in less than a month, hopefully. I'm very busy at my (real) work up to mid-march.

no patreon. I applied to github sponsor but the validation is in process.

LaserBee, 2020-02-26:
That's brilliant. Thank you again for your excellent work here!

supermerill, 2020-06-24:
i'll close it for now, if your concern is still valid, please re-open it.

closed 2020-06-24 by supermerill