supermerill/SuperSlicer #133: Internal error: <unspecified file>(299): duplicate key name

neslekkim, 2019-12-21:
### Version
Slic3r++_2.2.46.0_win64_20191212

### Operating system type + version
win10

### Behavior
crashes on first startup, and after starting again I get this when trying to use the configuration wizard
15:53:45: Internal error: <unspecified file>(299): duplicate key name
15:53:45: Internal error: <unspecified file>(299): duplicate key name

tonn0297, 2019-12-21:
i have same error.
v2.1.45.0 works fine. 
its always when you open the configuration wizard

supermerill, 2019-12-22:
I can reproduce if launched in an unconfigured environment (like when using the local_setting.bat).
I'll add it to my check list for creating a release.

audiorazor, 2019-12-26:
Came here to post the same issue.  Having the same problem.

neslekkim, 2019-12-26:
If I took an older version and run that, and then run the new version, it worked fined. So I guess it's missing some files.
I compared the slic3r++ files under profile\roaming, and the vendor folder is empty on the version that crash, but filled in the other folder.
The cache folder also only have idx file, not the ini file.

RomanTheLegend, 2019-12-27:
Same issue on Win7 x64 - freshly unpacked bins, no old configuration

Not so much info in logs:

```
Faulting application name: slic3r++.exe, version: 2.2.46.0, time stamp: 0x5df272ff
Faulting module name: slic3r.dll, version: 0.0.0.0, time stamp: 0x5df272f9
Exception code: 0xc0000005
Fault offset: 0x00000000007c76db
Faulting process id: 0x48f8
Faulting application start time: 0x01d5bcd6d8c5e7fd
Faulting application path: C:\Dev\3D_print\Slic3r++_2.2.46.0_win64_20191212\slic3r++.exe
Faulting module path: C:\Dev\3D_print\Slic3r++_2.2.46.0_win64_20191212\slic3r.dll
Report Id: 3d6be31e-28ca-11ea-bf3e-2c44fd657aa4
```

photog0411, 2020-01-01:
Opening the slic3r a second time, no more error

closed 2020-01-15 by supermerill