supermerill/SuperSlicer #134: Mac Version crashes when trying to import printer profiles or add a new printer

audiorazor, 2019-12-26:
### Version Latest built for MacOS

### Operating system type + version
MacOS 10.15.2

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
When I try to import stock prusa profiles or add a new printer it crashes.  I have attached a picture of the error it throws. 
[image]

audiorazor, 2019-12-26:
sorry, didnt realize this was already posted.

closed 2019-12-26 by audiorazor

supermerill, 2020-01-09:
just relaunch, it should resolve itself.
Or remove basic.ini and creality.ini from resources/profiles