supermerill/SuperSlicer #135: Travel moves past external perimeter with 0% fill (causing blobs)

espwarrior, 2019-12-27:
### Version

1.42.2 and 2.1.44

### Operating system type + version

Windows 10 Pro

### 3D printer brand / version + firmware version (if known)

CR-20 Pro and Duet Maestro RRF 2.05

### Behavior

I sliced the calibration cube from https://www.thingiverse.com/thing:3670991 at 0.2 layer height, 2 perimeters, 0% infill, 0 top layers, 0 bottom layers, 30mm/s inner and outer perimeter speed, 100% flow rate/extrusion factor, extrusion width = 0.40. When printing, large blobs at the seam appear. Closer inspection of the sliced model revealed that there are travel moves that extend past the outside of the external perimeter. This happens as soon as infill is missing.

[image]

When loading the exact same 3MF in PrusaSlicer 2.1.0, these travel moves are not generated.

[image]

Granted, there should be a number of mitigating settings (such as retract at layer change) and I don't use pressure advance yet, so I could probably minimize these blobs on the outside of my print. However, I'm also curious why I'm seeing this difference in slicing approach. I also couldn't find an option to influence this. Better to ask than to be sorry 😉 

#### Project File (.3MF) where problem occurs

Please see attached zipped 3MF file.

[Extrusion Factor Test Print.zip](https://github.com/supermerill/Slic3r/files/4006425/Extrusion.Factor.Test.Print.zip)

supermerill, 2020-01-02:
Found. It's a section I needed to review from a long time. Basically, it incorrectly thinks that the external perimeter around a hole instead of around the part.

espwarrior, 2020-01-03:
Thanks for the explanation and I'm happy that you found the cause.
And I'm grateful for your awesome work 👍

closed 2020-01-15 by supermerill