supermerill/SuperSlicer #138: Slicing error/difference  with 2.2.47 and 2.2.0_alpha

neophyl, 2020-01-11:
### Version
2.2.47.0 UI edition

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
Sliced for a modified cr10 with direct drive and 0.6 nozzle

### Behavior
I was using the latest ++ release to slice a project and give it a bit of a test at the same time.  The same project sliced fine in the previous prusa slicer releases (and also slices fine in 2.2.0 alpha2).

With 2.2.47 though there was a stl file that I just couldn't get to slice properly at all.  There's bits missing, I know this is usually down to bad model geometry but no errors showing for the part.  Also the prusa version slices it no problem so I don't think its the model.

Prusa 2.2.0 alpha 2
[image]

Slicer ++ 2.2.47.0
[image]

I have tried thin wall detection on/off, with values going down to 10%.  Tried adjusting perimeter sizes etc but nothing I have tried gets it to slice properly.

I saved the project with just the problem part  and confirmed that it still had problems with just that part used.  Project attached.  Hopefully someone can figure out why it has a problem.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[catch_test.3mf.zip](https://github.com/supermerill/Slic3r/files/4049830/catch_test.3mf.zip)
(just renamed it with zip on the end)

supermerill, 2020-01-12:
It's a bug in the thin wall algorithm, maybe.
To return to a good (prusa-like) behavior, enable "overlapping perimeters".

neophyl, 2020-02-03:
As this is now fixed I guess I should close ti :)

closed 2020-02-03 by neophyl