supermerill/SuperSlicer #140: anchor the first top solid layer

supermerill, 2020-01-13:
like https://3dprintingindustry.com/wp-content/uploads/2019/02/pathio-vs-old-slicers-768x1099.jpg
Add a setting to activate
if more than 1/2 ? perimeter(s) at this position, if this perimeter intersect with a solid bridged layer, then don't print this perimeter

JS3910, 2020-01-18:
That looks good, but why in this illustration are only the layers at the middle height inset into the perimeters? Should it not also apply to the topmost one?

supermerill, 2020-01-18:
i think it should.

JS3910, 2020-01-19:
Yep, I don't understand any conceptual difference between the first top layer on the two "side" parts vs the topmost bit, but that's on the original source for the image of course. I'm assuming that these are implied to be bridges going into the plane of the screen or have support not shown in the image, or this would not work so well at all :D.