supermerill/SuperSlicer #141: XY compensation for first layer, empty G1 codes

albert2004, 2020-01-13:
### Version
2.2.47.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
No matter

### Behavior
* slice 3 small objects

#### Project File (.3MF) where problem occurs
Issue 1:
First layer compensation (-0.2mm) does not work for thin walls. First layer is exactly this same as second one. In compare to PrusaSlices 2.2.0 alpha2, there is smaller first layer (and consistent according to new feature to adaptive compensation.
Supermerill version:
[image]
[image]
PrusaSlicer version:
[image]
[image]

Issue 2:
In generated gcode, there are existing empty G1 codes (not existing in PrusaSlicer, 
see lines 800-900), like:

G1 X23.771 Y35.780 E0.00037
G1
G1 X23.752 Y35.825 E0.00057
G1
G1 X23.734 Y35.870 E0.00077
G1
G1 X23.691 Y35.955 E0.00189
G1
G1 X23.667 Y35.995 E0.00130
G1 X23.603 Y35.958 E0.00205
G1
G1 X23.539 Y35.922 E0.00158
G1 X23.053 Y35.417 E0.01506

Attached project and output gcodes from Supermerill and PrusaSlicer:
[Issue.zip](https://github.com/supermerill/Slic3r/files/4052401/Issue.zip)

albert2004, 2020-01-22:
for XY compensation creates compilation error on linux / gcc 9.2.
In provided fix, you changed 2 functions:
ExPolygons _offset_holes(double hole_delta, const ExPolygons &input) const;
ExPolygons _smooth_curves(const ExPolygons &input, const PrintRegionConfig &conf) const;
in both of them, return value is specified under "if" and gc

albert2004, 2020-01-22:
for XY compensation creates compilation error on linux / gcc 9.2.
/var/tmp/portage/media-gfx/slic3r-supermerill-9999/work/slic3r-supermerill-9999/src/libslic3r/PrintObject.cpp:2387:1: error: control reaches end of non-void function [-Werror=return-type]
/var/tmp/portage/media-gfx/slic3r-supermerill-9999/work/slic3r-supermerill-9999/src/libslic3r/PrintObject.cpp:2496:1: error: control reaches end of non-void function [-Werror=return-type]

supermerill, 2020-01-23:
I corrected it in the branch 'stable'. I didn't merged it with master yet.

closed 2020-02-05 by supermerill