supermerill/SuperSlicer #142: 0° Bridge angle does not align bridging to axis.

BlazingBlackBeard, 2020-01-13:
### Version
2.2.47
### Operating system type + version
Windows 10 Pro 64 BIT

### 3D printer brand / version + firmware version
N/A

### Behavior
Configuring  a 0° angle for bridging angle does not align bridging to an axis, it will just default to the infill angle of 45° and try to bridge diagonally, not using the shortest path across gaps.

Reproducing this problem is easy, just load a model with a long and narrow bridging area, like the Skelestruder Mosquito mount and when it's sliced at an angle of 0° bridging, it will show in all it's glory.

#### Project File (.3MF) where problem occurs
_Upload a Slic3r++ Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r++, Slic3r PE 1.42.0-alpha and newer)_

kwaad2, 2020-01-13:
Disabling "Only one perimeter on Top surfaces" is the solution, so I would assume the issue lies within that.

BlazingBlackBeard, 2020-01-13:
> Disabling "Only one perimeter on Top surfaces" is the solution, so I would assume the issue lies within that.

I'm not sure anyone would agree that disabling one unrelated feature should be neccesary to have bridging settings work as intended.

This is xlearly a bug that needs fixing properly so that bridging angles can be set independently of other unrelated settings, not hacked or worked-around just to make it behave as intended at the expense of other unrelated features.

Granted, as a temporary fix people could disable single perimeter tops but this should only be temporary and a proper fix implemented so that unrelated functions work as intended, in combination. I've not tried it yet so I don't know whether it actually allows bridging to be properly set. These options should not affect eachother.

supermerill, 2020-01-13:
Currently, bridging angle is given by an obscure algorithm that tries to simulate different configuration and use the one that seems the best.
It's not an easy task, as I need to identify what's wanted (the conf with most short lines?, the one with the less long one?), and how to get it. So, sadly it may take some time.

For the "bridging angle", i have to look at what uses it.

supermerill, 2020-01-13:
For this particular piece:
* a volume in the layer at 6.44/6.60 is only composed of 2 layers. So with "only one perimeter on top", some perimeters are pushed back as overhang, and it's bad.
* using "no perimeters on bridge" -> "keep only bridges" resolve this problem (it's not perfect, but it's much much better). I didn't see any other place where this setting will create problem, so i didn't put it in a modifier mesh.

[SP_cageB_mosquito.3mf.v2.zip](https://github.com/supermerill/Slic3r/files/4055488/SP_cageB_mosquito.3mf.v2.zip)

kwaad2, 2020-01-14:
I would also like to point out that in Slic3r a 0 degrees for bridging is the default for "auto-generate" If you use 180, it will force generate the supports in the direction you want.
[image]

supermerill, 2020-01-14:
For the next version, the "bridging angle" setting will be enforced if not 0 (instead of being used as an default value for the algorithm).

supermerill, 2020-01-15:
With latest release
[image]
[SP_cageB_mosquito_v3.3mf.zip](https://github.com/supermerill/Slic3r/files/4066321/SP_cageB_mosquito_v3.3mf.zip)

closed 2020-02-05 by supermerill