supermerill/SuperSlicer #143: Support enforcers do not work and are bugged.

BlazingBlackBeard, 2020-01-13:
### Version
2.2.47

### Operating system type + version
Windows 10 Pro 64 BIT

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
When adding support enforcers under bridges, with the 'Don't support bridges' option deselected, supports are not generated under the part. However, in PrusaSlicer support enforcers are correctly generated for the same part and enforcer.

[image]

[image]

#### Project File (.3MF) where problem occurs
[SP_cageB_mosquito.3mf.txt](https://github.com/supermerill/Slic3r/files/4053177/SP_cageB_mosquito.3mf.txt)

Example of PrusaSlicer correctly applying support enforcers:

[image]

[image]

closed 2020-01-13 by BlazingBlackBeard

supermerill, 2020-01-13:
it's resolved?

BlazingBlackBeard, 2020-01-13:
Settings oversight, no problem here.