supermerill/SuperSlicer #144: first layer Z value Bug (2.2.47)

formadi, 2020-01-13:
### Version
2.2.47

### Operating system type + version
macOS 10.15.3 beta

### 3D printer brand / version + firmware version (if known)
prusa mk3

### Behavior
If you check the Gcode after creating the gcode, the Z value of the first layer is set to the Nozzle Size value instead of the Base layer Height value.

supermerill, 2020-01-13:
I need your .3mf to check, can you "save as" and copy-paste it here zipped (or renamed) ?

formadi, 2020-01-17:
> I need your .3mf to check, can you "save as" and copy-paste it here zipped (or renamed) ?

I checked a bit late. Attach the .3mf file.
[cube.3mf.zip](https://github.com/supermerill/Slic3r/files/4074732/cube.3mf.zip)

supermerill, 2020-01-17:
Your settings have 
`first_layer_height = 100%`
In a previous version, the % was in function of the base height, but it was a mistake, and changed it to be a % of the nozzle size.
You have to change it to 50-75%. GO into expert mode ->slicing -> first layer height.
If you use an old config from the wizard, you should update it. Replace the .ini from c:/users/myname/appdata/roaming/slic3++/vendor/ by the one(s) in install_dir/slic3++/resources/profiles

Now i understand why it's important to increase the version of profiles... i'll do it in the future.

Happy printing and sorry for this inconvenience.

closed 2020-01-17 by supermerill

formadi, 2020-01-17:
Ah ~ I understand now. I wasn't setting it up correctly. 
Thank you for a detailed description. 
and this is the Best Slicer I've ever used. ^^