supermerill/SuperSlicer #146: Supporting dense layer infill doesn't succesfully anchor on low infill percentage.

JS3910, 2020-01-18:
### Version
The Mac version posted last in issue 129, says version 2.2.47+UNKNOWN in file info. The about box is empty.

### Operating system type + version
MacOS 10.14.5

### Behavior
When infill is set to a low percentage, here 10%, the anchoring doesn't make sure that it actually extends to the nearest infill line, it seems to just be a set distance around the hole. Expected behavior would be that it extends to a place where the lines would actually be anchored to.

[image]
[image]
[image]

[Laser Camera Base.3mf.zip](https://github.com/supermerill/Slic3r/files/4081270/Laser.Camera.Base.3mf.zip)

supermerill, 2020-01-18:
Look at the algorithm it uses and the tooltip.

'Anchor' means 'enlarged by the anchor setting you see below'
'Automatic' means 'be sure to not turn where there are no perimeter'.

Maybe i should rename it...

The best you can do, imho is to increase the parameter on this layer (see 
[Laser Camera Base v2.3mf.zip](https://github.com/supermerill/Slic3r/files/4081733/Laser.Camera.Base.v2.3mf.zip) )
Or you can change it to automatic, but it will create much more "dense support", to be safe.

[image]

JS3910, 2020-01-18:
Oh, yes that was not obvious from the available options! The other settings generate this big sort of big grid pattern so "automatic" I presumed would just make it ... good enough that the solid infill layer can be printed succesfully :).

closed 2020-01-18 by supermerill