supermerill/SuperSlicer #149: Perimeters missing/Bridge gap from sliced models

vi-kings, 2020-01-19:
### Version
2.2.47.1

### Operating system type + version
Windows 10 Jan 2020 Update

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + 3.8.1 fw

### Behavior
1. Import the 100mm bridging test from here: https://www.thingiverse.com/thing:357674/comments or import the attached 3mf
2. Slice with any profile for example: https://prusaslicerprofiles.com/download/pretty-bmg-pla-v3-1-5/
3. See that the mm is incomplete.
4. Also see that the Bridge has a gap between the Overhang perimeter and the Bridge infill

slicer++ top and bottom:

[image]
[image]

PrusaSlicer with the same profile:

[image]
[image]

#### Project File (.3MF) where problem occurs
[Bridging_100mm.zip](https://github.com/supermerill/Slic3r/files/4083258/Bridging_100mm.zip)

Hope this helps in fixing it, please let me know if I did something wrong or if I can verify something.

supermerill, 2020-02-05:
With the prusaslicer version, the two external "pastel  blue" brige are well inside the "bright blue" perimeter bridge. I don't think that's a good idea.
If you want to fill it more, you can set the bridge overlap to 120%.

I really don't know how to improve on this. Maybe with thinner bridges... if it's even possible... But the algorithm to decide that may be horribly complex to create.

Zulaner, 2020-06-03:
[image]
[image]

supermerill, 2020-06-03:
@Zulaner, can you post your .3mf? (project->saveas, then zip it and drop here)

Zulaner, 2020-06-03:
here goes the file
[Duct_Led_V3_Cortes.zip](https://github.com/supermerill/SuperSlicer/files/4722679/Duct_Led_V3_Cortes.zip)

supermerill, 2020-06-03:
@Zulaner it's not a bridge per see but the solid infill from the top shell that is printed with bridge flow. You have to put some infill to support it (or use dense infill, but it does not do a good job with this object).

closed 2021-11-19 by supermerill