supermerill/SuperSlicer #151: Request: Gradient/Variable infill

vi-kings, 2020-01-19:
### Version
2.2.47.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + 3.8.1 fw

### Behaviour

This is a feature request to implement gradient infill as described here:

https://www.cnckitchen.com/blog/gradient-infill-for-3d-prints

It seems to have superior mechanical properties in many situation and I think it would be a nice addition.

Someone has also already implemented it into a fork of PrusaSlicer so it might be a merge job :)

Found here: https://github.com/thenickdude/PrusaSlicer/tree/variable-width-infill

supermerill, 2020-01-19:
I didn't see in the video where it show any "superior mechanical properties".
There is really?

vi-kings, 2020-01-19:
Around the 10:16 minute mark you can see it does have higher Stiffness rating compared to the 30 or 46% infill. True, there isn't much difference with the hook test, but for applications that would need higher stiffness it could benefit from it.
Can even give it a fancy name like Bone Marrow infill.

supermerill, 2020-01-20:
That comparison isn't fair. Grandient infill is the same as adding a perimeter and removing some infill, but it the video, the comparison pieces have the same number of perimeters.
I'm sure if you use decrease infill and perimeters, it will be  stronger than gradient infill, as in my tests the strongest settings where with 0% infill /max perimeters (for bending test).

And comparison must be at the same weight, and i don't think it's the case in this video.

I'll implement it if i can see a benefit. My own testing say that there isn't, but if you can provide me some more comprehensive testing (same weight, different perimeter count, different infill pattern), then i will gladly add it to slic3r.
Prove me that i'm wrong!

vi-kings, 2020-01-20:
I am afraid I don't have the capacity to build a sufficient test stand to prove this and I pull my own information from the article and the linked video. I am not the author of the article above, just found and I believe it is pretty neat.

Cura also can do this already in post-processing and zmorph apparently also does something similar to save material: https://support.zmorph3d.com/hc/en-us/articles/115002839493-Adaptive-Infill

If you don't believe there is any benefit to this, then feel free to close this and I apologise for have wasted your time.

fiveangle, 2020-10-14:
I agree with you @supermerill that for generic spindly items such as CNCKitchen's standard test pieces, the impact appears minimal so probably not worth maintaining just for that.  However, for voluminous objects requiring object shear strength, I have found it to be quite an improvement.  I have not done extensive testing, but in this simple cone design intended to support the 52lb weight of my winter wheels/tires suspended from the ceiling of my garage on threaded rods, it required me adding 2x perimeters and increasing infill density approx 5% in order to get them to not fail with 2x 90lb cement bags + ~10lbs of rope/straps before they would not shear off the end of the threaded rod when I added my own weight (approx 175lbs).

However, when I printed the same item using 50-250%, 10cm variable infill i was able to maintain the standard "0.3mm DRAFT" parameters of 2 perimeters, 20% infill density, and it would not break when I added my own weight to the 2x 90lb cement bags + ~5lbs straps/etc.  Although print time is very similar, the increased structure for the outside and inside load-bearing surfaces is dramatically increased far beyond what traditional settings can accomplish.

Increasing infill width directly in the extrusion settings is another option that would accomplish similar strength, but would require the user to calculate the best infill for the model by translating the extrusion width in their head.  With the variable width infill option, it offers both a very quick and easy method for accomplishing this directly within the infill section, plus does so with less actual plastic used.

This is just one case where the feature improves, but I suspect there are many others.  One I have been meaning to test is to set the low variable to "0%" so that the model includes only infill directly attached to the perimeters, while the rest of the model is hollow.  This could dramatically improve print speed for some models that require only perimeter rigidity but not overall model rigidity, and dramatically reducing print time.  Adding a short 0-250% 3mm variable infill I suspect could allow rigidity similar to adding 3-4 perimeters.  Again, I don't believe the feature is a drop-dead necessity as one can hack the slicer to obtain similar results, but it does show promise for many voluminous functional parts and other outlier uses to potentially improve strength for the same material used, or reduce material used to obtain similar strength.

[image]

[image]

[image]

[image]

supermerill, 2020-10-14:
Just a note that prusa added an "Adaptive Cubic" and a "Support Cubic" infill pattern for 2.3, so that will be merged.