supermerill/SuperSlicer #152: Feature request: manual seam placement tool

lokus77, 2020-01-21:
Hi supermerill. I love Slic3r++, very nice job!!  
I know slicer++ has "hidden seam" option, but.. 

Maybe would be nice (and maybe unique feature, does not exist in any slicer) if the
user could **"paint" in 3d view the seam position** (across Z ). So user can decide where goes seam in each part of a model.  Would be something like variable layer height editor. Not sure if is easy to implement.

I did prusa team request https://github.com/prusa3d/PrusaSlicer/issues/2527
and several users liked it. 

Time ago, some users tried similar solutions:
https://github.com/prusa3d/PrusaSlicer/pull/728
https://github.com/prusa3d/PrusaSlicer/issues/342

supermerill, 2020-01-21:
Maybe a modifier mesh ?
The seam will go at the nearest point, relative to the modifier mesh (at the same Z).
So you can add a cylinder for simple placement (like the "rear" option.
And you can create a complex mesh if needed.

wesc23, 2020-06-13:
Another option would be Kiss like seam settings.  In Kiss you specify a sector where kiss is allowed to place seams.  Allows you to dither a seam on the back of a part, for example.

stevenignys, 2020-06-14:
Hi, This would be very useful. I could see it working like the support enforcer and support blocker.

Or maybe could use the same UI element as the Place on Face tool and allow you to select faces you dont want the seam on or do for instance?

Stephan3, 2020-07-21:
+1 for kiss seaming
[image]

dartrax, 2020-07-26:
> +1 for kiss seaming
> [image]

@Stephan3, has this actually something to do with where on the outside perimeter the seam should be placed? It looks like the seam gets hidden inside by some perimeter-x-crossing. If that's the case, I would recommend openig a new feature request and explain it. Looks very interesting!

dartrax, 2020-07-26:
> Another option would be Kiss like seam settings. In Kiss you specify a sector where kiss is allowed to place seams. Allows you to dither a seam on the back of a part, for example.

I was not able to place a seam at the inside of a C-Shaped object with kisslicers way: https://github.com/prusa3d/PrusaSlicer/pull/728#issuecomment-583409225

supermerill, 2020-08-19:
Current implementation:
 - you can add sphere via the contextual menu
 - for each layer, it will get the nearest sphere in the Z axis
 - it will put the seam at the nearest point from the center of the sphere (in the xy plane)

Maybe later I or someone can modify the slicer to also slice the seam object to be able to have a complex seam object.

closed 2020-10-22 by supermerill