supermerill/SuperSlicer #153: Material Interface for supports when set to Ironing results in error

Spike718, 2020-01-22:
### Version
2.2.47

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Mk3s

### Behavior
Error is thrown during Slicing with Supports which use Ironing interface pattern.
Error: "Flow::mm2_per_mm() produced negative flow. Did you set some extrusion width too small?"
Using a default profile for a .4 nozzle, this issues occurs when generate support is enabled and the Pattern for support material interface is set to Ironing. Other interface patterns work fine. 

Enabling ironing in the infill section does not generate this error if the Ironing interface pattern is not selected for the Interface Pattern in the Support Material area. This was to verify if the infill setting elsewhere would cause errors however as noted this did not occur.

Steps:
Place an STL that will need supports
Enable Supports
Go to Print Settings ->Support Material
For the Pattern under Options for Support Material Interface, select Ironing.
Slice
Expected Result: Should slice without error and produce an interface with an ironed (ironing) pattern

Actual Result: Error will appear.

_Is this a new feature request?_

supermerill, 2020-01-22:
can I have your project file? menu > save as -> rename/zip -> drop it here.

Spike718, 2020-01-22:
Of course! Attached

[Ironing_Interface_issue153__.zip](https://github.com/supermerill/Slic3r/files/4098520/Ironing_Interface_issue153__.zip)

closed 2020-04-21 by supermerill