supermerill/SuperSlicer #154: [Feature Idea] Allow PrusaSlicer Macro Code in filament start/end gcode

toskium, 2020-02-02:
### Version
2.2.47

According to this github wiki:
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

There is a bit of a macro language available in PrusaSlicer, ergo it should be available in Slic3++.

I just recently discovered that the macro code doesn't work within the filament custom start/end gcode (slic3r++ displays an error message) it only works in the start/end gcode of printer settings.

However, it would be really useful in the filament start gcode as well. 
For example: if a specific filament needs more flow or less retract only for the first layer, this could easily be tuned.

supermerill, 2020-02-03:
You have access to "current extruder" but not the layer height.
If you want a script to be executed at each layer change, write it in printer -> gcode -> after layer change G-code.
exemple: 
`{if filament_type[current_extruder]=~/.*PLA*/}PLA{else}PAPLA{endif}`

note that you can access all filament properties this way (by accessing them with current_extruder)
If you can use multiple filament, you should also copy/paste your script in "toolchange gcode"

toskium, 2020-03-07:
The fact that the layer ID is not available in the filament start-gcode was the whole idea behind the feature request. 
Let me try to explain: I am using firmware retraction, and some of my filaments do need a smaller retract value to stick well to the print surface. So the start gcode would look something like this:

{if layer_z < 2}M207 S5.2 F6000 Z0.3{elsif layer_z >1}M207 S5.6 F6000 Z0.3{endif}

If I put this into the printer config I would have to create many printer configs one for each filament that needs different retraction values.

Imho it would make total sense to make the layer_z variable available in the filament gcode.

supermerill, 2020-06-20:
seems like i added layer_num and layer_z since 11 mars. so it should be in the latest release.

closed 2020-06-24 by supermerill

toskium, 2020-07-10:
Sorry for being unresponsive, didn't have the time to test your fix.
I am using Superslicer version 2.2.52 Win64 and from my tests yesterday the feature is not yet working.

The error message is gone when trying to slice the gcode with macro code in the filamnet custom gcode box but it does not reflect the intended changes to the gcode.

Here is what I did try yesterday:

I added this to the custom filament gcode:
`{if layer_z< 0.4}M207 S5.2 F6000 Z0.3{elsif layer_z > 0.2}M207 S5.8 F6000 Z0.3{endif}`

Expectation 1: no slicing error, works!
Expectation 2: modified M207 gcode at each layer change, does not work. 

This is what get's written to the gcode:
`; start_filament_gcode = "M572 D0 S0\nM207 S5.7 F6000 Z0.3\n;M207 S5.8 F6000 Z0.3\n\n{if layer_z< 0.4}M207 S5.2 F6000 Z0.3{elsif layer_z > 0.2}M207 S5.8 F6000 Z0.3{endif}"
`

What should have happened was that this is reflected in the layer change gcodes that are usually defined in the printer settings.
If I am adding the same macro to the "Before layer change G-Code" box the gcode gets properly modified to look like so:

```
;LAYER:0

M207 S5.2 F6000 Z0.3
.....
;LAYER:1

M207 S5.8 F6000 Z0.3
.....
;LAYER:2

M207 S5.8 F6000 Z0.3
```