supermerill/SuperSlicer #155: Feature request: compatibility with macOS Catalina

liftbag, 2020-02-02:
### Version
_2.2.47 and newer_

### Operating system type + version
_macOS Catalina 10.15.3 (19D76)_

### 3D printer brand / version + firmware version (if known)
_irrelevant_

### Request:
__I wonder if it was possible to make the untested macOS builds compatible with macOS Catalina 10.15.
I would be really happy to replace PrusaSlicer with Slic3r ++, as I see options that have been missing for too long in PS. But my work environment is Catalina and, perhaps also because of my poor skills, I was unable to compile from the sources.__

liftbag, 2020-02-03:
I managed to start slic3r++ on Catalina!
The steps I followed:
- Installed on Mojave
- Fixed icon display
- I started the application and configured it
- Closed Mojave I started Catalina
- Copied the executable from Mojave's Applications folder to the Catalina Applications folder.
At this point the program started normally.

liftbag, 2020-02-03:
It seem to be a permission issue.

[image]

supermerill, 2020-02-03:
I'll add a chmod in the build script.

How did you fix the icon display?
what's the "configuration" you did?

liftbag, 2020-02-03:
> I'll add a chmod in the build script.
> 
> How did you fix the icon display?
> 

For some reason the slic3r.icns icon in Contents/resources is not displayed in the app launcher.
Here you can see how it appears (I renametd it slic3r+++ to allow parallel installation)

[image]

liftbag, 2020-02-03:
> what's the "configuration" you did?
>

The normal configuration wizard on first startup.

liftbag, 2020-02-03:
Here the differences of permissions between the application imported from Mojave and the one installed directly on Catalina. The executable flag appears to be missing.

[image]

supermerill, 2020-02-03:
Oh, it's the zip operation that brake it. I'll do something.

liftbag, 2020-02-03:
Thanks for your time. I am getting familiar with this PrusaSlicer on steroids and I like it a lot.
Thank you so much for your great job!

liftbag, 2020-02-03:
Actually
`chmod +x <path_to_Slic3r++>/Contents/MacOS/slic3r++`
do the job.

supermerill, 2020-02-03:
can you test the dmg.zip here?
https://github.com/supermerill/Slic3r/actions/runs/34323886

and the tar.zip if it's not working as intended.

liftbag, 2020-02-03:
> can you test the dmg.zip here?
> https://github.com/supermerill/Slic3r/actions/runs/34323886
> 
> and the tar.zip if it's not working as intended.
>

Both work perfectly! Only the icon is missing, but it's not a problem.

closed 2020-02-03 by supermerill

antst, 2020-02-15:
Just in case, as it was here, icon issue on Catalana is related to wrong case
Contents/resources/slic3r.icns must be Contents/resources/Slic3r.icns