supermerill/SuperSlicer #157: [Feature Idea] a setting to blip fan up to speed, some time before it actually needs to come to speed

Photogad, 2020-02-04:
### Version
any

### Operating system type + version
any

### 3D printer brand / version + firmware version (if known)
mk3s

### New Feature Request:
I have noticed sometimes that really my print fan does not come up to full speed before it starts bridging, on certain bridges this is an issue and I think the bridges sag more than they need to. In particular this is problem with PETG where I print with no fan at all, and only use fan for bridging or overhangs. So usually the fan is running at 0% speed (off) but it does not reach 100% speed before the bridging starts.

I would like to see a setting that could be toggled into slic3r where I could specify how many miliseconds in advance the print fan should start turning on, BEFORE a print move that requires fan is called for. So like for example I could set it to turn on "500ms" before the actual time that it is ever needed.

If you think this is bad idea or can't be done, feel free to close the issue. I am request it here because Prusa team do not usually listen to feature request and I think there is more chance merril can add it.

_Is this a new feature request?_
yes

supermerill, 2020-02-04:
printer -> fan startup delay
put 0.5 or -0.5

it also add a 'd' parameter to let the firmware know when it should be up to speed, but no firmware support it yet. If this parameter makes your firmware to bug, add a '-' before the number.

closed 2020-02-04 by supermerill

Photogad, 2020-02-05:
> 
> 
> printer -> fan startup delay
> put 0.5 or -0.5
> 
> it also add a 'd' parameter to let the firmware know when it should be up to speed, but no firmware support it yet. If this parameter makes your firmware to bug, add a '-' before the number.

thanks! I missed it because this setting in a weird location, but i guess it makes sense under firmware.

just to clarify, if I set it to  -1.00 it would make the fan startup 1 second sooner than needed, and  1.00 would make the fan startup 1 second later than needed? I don't understand also the D parameter

supermerill, 2020-02-05:
if I set it to -1.00 it would make the fan startup 1 second sooner than needed, 
and 1.00 would make the fan startup 1 second sooner than needed.
The D parameter ios only set if >0, it's the number of ms that the printer has to spin-up the fan. Because It's not possible to put the GCODE exactly at 1s before (it's at least 1 s before). The D parameter tells you by how much we are soon.