supermerill/SuperSlicer #159: Graphical glitches in seam

guestisp, 2020-02-05:
### Version
2.2.47

### Operating system type + version

```
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan
```

### Behavior

As visibile in the screenshot (i've also atteched a 3mf file), there are some ugly glitches on the seam location (I think it's the seam location)

[image]

[image]

_Is this a new feature request?_

no

#### Project File (.3MF) where problem occurs

[cube-40-40-40.zip](https://github.com/supermerill/Slic3r/files/4162193/cube-40-40-40.zip)

supermerill, 2020-02-05:
I forgot to auto-uncheck "vase mode seam" when "external periemeter first" is unchecked.
To correct your issue, check "external periemeter first" , uncheck "in vase mode" then  uncheck "external periemeter first".

guestisp, 2020-02-05:
Your are right, manually unchecking that solve this issue

supermerill, 2020-07-03:
forgot to say it's corrected now

closed 2020-07-03 by supermerill