supermerill/SuperSlicer #160: Hightlight seam location in preview

guestisp, 2020-02-05:
### Version
2.2.47

### Operating system type + version

```
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan
```

### Behavior

Please add a "seam" option in preview, to show the seam location, like "Travel", "Retractions", "Unretraction", "Shells", "Legend" and ...... "Seam"

_Is this a new feature request?_

yes

supermerill, 2020-02-05:
why not. I'll look if it's easy to add.

guestisp, 2020-04-29:
Did you have a chance to check this ? It's very useful.
Also, I would suggest to differentiate retraction and unretraction colors from the preview. Actually they are both white.

supermerill, 2020-04-29:
noted.
Didn't took the time for this yet.

wesc23, 2020-06-13:
+1 on this feature.  visualizing seam placement is very important to reduce reprints due to bad seem locations.

stevenignys, 2020-06-14:
+1 this would be very useful. Also a way to specify the seam or allowable location maybe like the support allow and stop blocks

supermerill, 2020-06-14:
> Also a way to specify the seam or allowable location maybe like the support allow and stop blocks

you can discuss on that in #152

supermerill, 2021-06-02:
implmeented by prusaslicer

closed 2021-06-02 by supermerill