supermerill/SuperSlicer #161: Alternative preview color schema

guestisp, 2020-02-05:
### Version
2.2.47

### Operating system type + version

```
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan
```

### Behavior

on some environment (old monitors and so on), the current preview color schema is too dark and with low contrast. Internal and external perimeters could be confused, as one is orange, the other is yellow, but both very dark .

The same is for internal, solid, topsolid and bridge infill. Dark red, purple, red and violet.
And so on....

An alternative schema could be added, with much brighter and contrated colors, like S3D or Cura
Maybe reducing the darken effect would help ?

_Is this a new feature request?_

yes

supermerill, 2020-02-05:
these ones? 
[image]

guestisp, 2020-02-05:
Yes, i'm referring to that colors (obviously, any changes must be reflected on the plate preview)

supermerill, 2020-02-06:
here there are

```

    // sets colors for gcode preview extrusion roles
    std::vector<std::string> extrusion_roles_colors = {
        "Perimeter", "FFFF66",
        "External perimeter", "FFA500",
        "Overhang perimeter", "0000FF",
        "Internal infill", "B1302A",
        "Solid infill", "D732D7",
        "Top solid infill", "FF1A1A",
        "Bridge infill", "9999FF",
        "Gap fill", "FFFFFF",
        "Skirt", "845321",
        "Support material", "00FF00",
        "Support material interface", "008000",
        "Wipe tower", "B3E3AB",
        "Custom", "28CC94"
    };
```
I can't make them brighter, they already saturated.
Maybe it's more of a problem with opengl, and shaders that makes them looks dim on your computer.

Maybe you can post some screenshot, to explain a bit more the problem?

guestisp, 2020-02-06:
Last time i've checked, there was a function that makes everything darker, let me check...

guestisp, 2020-02-06:
IIRC this is where colors are made darker:
https://github.com/supermerill/Slic3r/blob/master/src/libslic3r/GCode/PreviewData.cpp#L338-L349

if you set something to white `#FFFFFF`, like gap fill, it will be rendered in light-gray. Light-gray is not white as `#FFFFFF`

On older monitor or low-contrast enviroment, everything is too dark and low constrasted, even the fan speed preview is hard to distinguish, almost the same blue scale.

Look at S3D preview, it's much much bright. Even Cura is much brighter

supermerill, 2020-02-06:
Tthis code just translate a color from 0-255 to 0-1. I can't see anything that can make it darker.

#FFFFF00 -> int -> [FF, FF, 00] -> [1.0, 1.0, 0.0]

supermerill, 2021-03-19:
If the problem appear again, please reopen.

closed 2021-03-19 by supermerill