supermerill/SuperSlicer #162: Feature request: temperature view in slice vieport

liftbag, 2020-02-06:
### Version
_next releases_

### Operating system type + version
_Any_

### 3D printer brand / version + firmware version (if known)
not relevant

### Request

Now that it's possible to simply set a multiple temperature print from the gui using the "Add custom g-code" option, it would be useful to be able to graphically display if there are temperature setting errors, in the same way as controlling the print speed , layer height, ventilation etc.
Currently the only way to do this, is to go through the inserted g-code point by point.

liftbag, 2020-02-06:
A classic example is setting up a temp tower. At each g-code entry window, the program proposes the last command entered. The temperature must be changed according to the established basic step, for example 5 decreasing degrees. If by distraction a command is duplicated rather than changed, with the temperature by color display this will be immediately evident.

liftbag, 2020-04-20:
@supermerill  do you think it is possible to include it in the next release?

supermerill, 2020-04-20:
I didn't began to work on this, and last time i added a "color" setting, it took a bit of time.
I'll push a new release before, as there are many bugfixes waiting to be published.

liftbag, 2020-08-21:
Thanks a lot, it seems perfect. Can be closed imo.

closed 2020-10-22 by supermerill