supermerill/SuperSlicer #163: Crash in slicing  with Keep only bridges

liftbag, 2020-02-07:
### Version
_2.2.47_2020-02-03_

### Operating system type + version
_macOS Catalina v.10.15.3_

### 3D printer brand / version + firmware version (if known)
_custom printer based on reprap firmware_

### Behavior
* _The macOS version of the program suffers from frequent crashes on my iMac Retina 5K, 27-inch, 2017, which I will document gradually. PrusaSlicer 2.2.0 alpha2 turns out to be much more stable at the moment_
* _Steps needed to reproduce the problem_
  * _Open the program, load the STL, set "Keep only bridges" on No perimeters on bridge areas, hit Slice now._
* _Expected Results: completion of the slicing action._
* _Actual Results: slic3r++ suddenly stops with the following warning window_
[image]

_This is the error message if the program is launched from the terminal_
[image]

_

_Is this a new feature request? No_

#### Project File (.3MF) where problem occurs
[Archivio.zip](https://github.com/supermerill/Slic3r/files/4170077/Archivio.zip)

liftbag, 2020-04-20:
If I use the attached 3mf, the problem exists even now, but if I try to recreate the issue from scratch and with a new profile, I have no problem. I think it is ultimately caused by some wrong profile setting.
I believe it can be closed.

supermerill, 2020-06-22:
seems to be fixed by something else

closed 2020-06-22 by supermerill