supermerill/SuperSlicer #165: Unpredictable results with interior brim option.

liftbag, 2020-02-10:
### Version
_2.2.47.1_

### Operating system type + version
_macOS Catalina 10.15.3_

### 3D printer brand / version + firmware version (if known)
_not relevant_

### Behavior
* _Generating brim inside holes doesn't seem to work, and "Interior brim width" doesn't seem to work well._
* _Steps needed to reproduce the problem_
  * _Place an object on the bed that has openings inside the bottom layer (I have attached an example step file)._
  * _Select "Brim inside holes" or enter a non-zero value in "Interior brim width"._
  * _Hit "Slice now""._

* _Expected Results_
  * _With the "Brim inside holes" option, the brim should be generated inside the holes with the same width as the external brim._
  * _With "Interior brim width" a brim of the desired width should be generated in the internal openings of the first layer._

* _Actual Results_
  * _The "Brim inside holes" function doesn't seem to work_
  * _"Interior brim width" function doesn't generate brim on all internal perimeters and it seems to work only in internal perimeters close to X = 0. Complex shapes seem ignored by the generation algorithm._

Here "Brim inside hole" is active
[image]

Below the "Interior brim width" option is set to 3mm. The object is presented rotated in different angles. Depending on the position, the brim is generated partially, always in different areas.
[image]

[image]

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Brim inside holes.zip](https://github.com/supermerill/Slic3r/files/4180023/Brim.inside.holes.zip)

supermerill, 2020-02-10:
thank you for the report, i'll look at it asap.

liftbag, 2020-03-28:
Just a mention, but the problem is still present inthe  version 2.2.48.1.

[image]

liftbag, 2020-03-28:
Oh, and it would be of great use, precisely in the printing of gears, to be able to insert only the brim inside the holes, so as to have the profile of the teeth clean, but at the same time to ensure a better adhesion of the lightened gears.
Here I tried to insert only brim into the holes, setting the brim width to zero and interior brim width to 4mm.

[image]

[image]

supermerill, 2020-03-28:
[image]

i wrote '=' instead of 'add' ...

liftbag, 2020-04-20:
Solved! Thanks @supermerill. Closing.

closed 2020-04-20 by liftbag